mrehkopf/sd2snes #167: SuperCIC Key on PIC12f629 working only as NTSC

edgarCarrasquilla, 2021-11-01:
Hi all, 

I'm having a strange issue with SuperCIC Key, programmed in a PIC12F629. I have programmed many of them in several and different roms, all PAL roms in JAP cartridges, always desoldering original CIC and soldering PIC12F629 following official schema, and soldering PAIR pin to VCC to deactivate PAIR mode, as my SNES is modded. 

Most of them works correctly: my modded PAL SNES detects them as PAL cartridges, but in some cases, it detects them as NTSC. For example: Terranigma Spanish version. Orogrammed this game as usual, soldering the PIC12f629 with last version of key .hex file, but my SNES detects it as NTSC. Powering up SNES as 50hz forced, it starts as 60hz and after 9 seconds turns to 50hz. In auto and 60hz forced modes, starts as 60hz. 

The problem with this game is that is software region protected, so although the rom is PAL (tested in emulator and in a everdrive), it starts as 60hz and software protection blocks it. 

At the moment, I have: 
- Double-checked all soldering points
- Assured that the PIC is labeled correctly (12F629)
- Programmed a new PIC and soldered again (same result) 
- Reseted several times and tried in all modes
- Tested many other games in which I used the same PICs and code, and worked correctly
- Tested other games for which I used the same PCB (1J3M-20), that also works correctly 

Any ideas? 

Thanks!

edgarCarrasquilla, 2021-11-02:
I have made another test: I used a PIC from another working multiregion cart (works as expected), desoldering from it and soldering to Terranigma PCB, with same result; in Terranigma PCB (1J3M-20), it works like a NTSC CIC always.

edgarCarrasquilla, 2021-11-07:
After some more (much more) trial and error, I think I found what my problem was. 

Supercic Key agree the region with Supercic Lock when I turn up my SNES. When, for the first time (a new cart with a supercic key recently programmed), as 60hz, both parts agree they are 60hz. If later I turn up my SNES in 50hz mode, as the PIC was previously set to 60hz, it detects it as so. 

So, my ultimate doubt is: is it possible to force a supercic key to change region on a supercic lock modded Super Nes?

mrehkopf, 2023-02-06:
That's true; there is no way to force the detected region when a SuperCIC communicates with a SuperCIC. A SuperCIC key powered cartridge will always adjust to the region of the last unmodded console it was connected to. The SuperCIC lock in turn will also adjust to the SuperCIC key's region. Thus, there is no actor governing the region.
Forcing the region *might* work if pair mode is allowed (SuperCIC key pin 4 to GND) and the CIC D0 an D1 lines are pulled to the desired levels (NTSC=0V; PAL=5V) with resistors (~4.7k-10k). In that case the SuperCIC lock would change its behaviour though and act more like a normal console (no switching using Reset button etc.); the cartridge would have total control over the region. Also it might trigger an error during authentication with an unmodded console (stock CIC) because of the resistors.

Maybe I should add a CIC key with a configurable region?