mrehkopf/sd2snes #169: RAD2X + FXPAK Pro video issue with firmware V1.11.0 BETA 1

ShiroTheSniper, 2021-11-19:
Hi! I would like to report an issue with the firmware v1.11.0 beta 1.
My setup:
SNES SNS-CPU-GPM-01
RAD2X (smoothing filter off)
FXPAK PRO (brand new)

Video issue:
[https://youtu.be/hEnnJrJCGu4](url)

I don't have the problem with these setups:
- RAD2X + firmware 1.10.3
or
- Standard AV cables + firmware v1.11.0 beta1
or
- RAD2X + Any real game carts

jon-nielsen, 2021-11-19:
I have a similar issue with a FXPAK Pro but only on my 2-chip SNSP-CPU-02.  On my 1-chip SNES mini there's no problem. The interesting thing is that a SD2SNES from AliExpress works just fine on both systems.

bkoropoff, 2022-01-08:
It looks like transient voltage drops due to the combined extra current drawn by the RAD2X and the FXPAK on the beta firmware.  Things you could look into:

1. Make sure you have an adequate power supply.  The OEM AC adapter is rated for 850mA, but you might want more than that since that rating wasn't chosen with flash carts and active AV cables in mind.
2. Replace the aging electrolytic capacitors in the SNES, particularly the main power smoothing cap(s).
3. Replace the old 7805 regulator, maybe with a 78S05 for the additional current capacity.