mrehkopf/sd2snes #171: FXPak Pro - Maerchen Adventure Cotton 100% won't run

pseiler, 2021-12-10:
Hi Ikari,

I just tried to play "Maerchen Adventure Cotton 100%" on my FXPak Pro and it just won't load.
I tried the SNES burn-in test and my PAL SNES with a SuperCIC seems fine. I also didn't face other issues. The diagnostics firmware (please merge the zip provided on your website as it took me way too long finding the version for the FXPak pro in the comments) passed all tests.

It seems that other people (maybe with the old FXPak model) don't have issues running the game (at first I tried a english translation romhack and people confirmed it worked on their system).
I'm on firmware 1.11.0beta but also tried 1.10.3 and both firmware won't run the game.

I haven't faced a single issue beside this one. I ran many romhacks and roms that matches the no-intro database without any issues.
I also ruled out a bad dump on this specific title.

Greetings from germany and keep on the good work!

symbios24, 2021-12-10:
Hi there try an older firmware until the game works finat

pseiler, 2021-12-10:
> Hi there try an older firmware until the game works finally

I have an FXpak pro. I cannot use firmwares older than 1.10.0.
I tried 1.10.1 - 1.11.0 beta and none of them works.

symbios24, 2021-12-11:
Did you try or you just saying??

pseiler, 2021-12-11:
I also found a comment on your website that mentioned this issue:
https://sd2snes.de/blog/archives/1076#comment-991899

I test if reseting helps.
Also I wanted to link the romhacking.net topic (maybe this helps too)
https://www.romhacking.net/forum/index.php?topic=33888

pseiler, 2021-12-15:
I now tried to reset a lot and didn't get it to run. I tried both. Resetting to menu and simply reset the game itself.
Was the same result.

mrehkopf, 2021-12-15:
Hm, strange. I could imagine that the game has some sort of timing-sensitive code regarding the APU (audio processor). The PAL SNES has a slightly different clock ratio of CPU:APU than the NTSC SNES so maybe it trips on that somehow. Does it make any difference if you run the game in 50Hz or 60Hz?
Have you tried running the "SNES Burn-In Test Cartridge Rev. D" to check for issues that your SNES itself might have? (Turn off in-game hooks before you run it or the HV Timer test will sometimes fail. ^^)

pseiler, 2021-12-15:
Thank you for the fast answer.

I haven't tried it in 50 Hz. Just used Auto setting from the SD2SNES superCIC settings. But I'll try. 
But regarding all other settings (I disabled everything else (hooks, cheats, auto-region-patching, etc...)

But yes I tried the burn-in test with the exact Rom you mentioned and found out to disable In-Game hooks for the HV Timer test.  Maybe I should've clearified it better in the start post. So the SNES itself works absolutely fine. 
But I have a friend who owns a SD2SNES and uses a SuperCIC PAL SNES and it runs fine for him (but I don't know which firmware, but it seems he uses 1.11.0 beta or the fork which enables SGB as he mentioned playing SGB with the SD2SNES).

pseiler, 2021-12-15:
@mrehkopf in 50 Hz mode it works flawlessly. Only when running in 60 Hz mode it won't start.
Having the menu in 60Hz or 50Hz doesn't affect it in any way. 

I ask my friend if he activated the SuperCIC in the SD2SNES and if the game runs on his SNES without an issue. 
Is this something thats fixable by the SD2SNES or would it also affect a real / repro cartridge?

mrehkopf, 2021-12-16:
I came to the same conclusion yesterday - it works in 50Hz mode but not in 60Hz mode except on a console with NTSC master clock (21.477MHz instead of 21.282MHz). It's very like some sort of race condition inside the game's code itself. Almost sounds like they got lucky it actually boots on NTSC consoles at all. :D 
I'd expect it to fail in the same way with a simple repro/ROM cartridge. I'm going to trace the game later using a logic analyzer to see where it hangs specifically and maybe put it on a repro cart to verify :)

pseiler, 2021-12-16:
Awesome. Thank you very much for your effort. I'm looking forward to your findings and news. I'm sorry that I'm not a capable of helping you as it seems quite a complicated task.

mrehkopf, 2022-02-01:
It boils down to the game using a buggy RAM clearing routine.
They forget to prepare a small portion of RAM, then subsequently expect certain values which aren't always there. The game then crashes a number of times before booting successfully (or sometimes doesn't). I found that it often manages to boot after 12 seconds of waiting or so.
Apparently in 50Hz mode the CPU is interrupted at different times while executing from undefined address space so it might recover faster because of that.
The "proper" fix would be to fix the game (it is a single byte patch - must have been a typo in the original code) but I will capture a bunch of RAM dumps after cold boot from a number of SNESes to mimic the initial state of original hardware better.

pseiler, 2022-02-01:
Thats awesome to hear. And I'm very thankful that you investigate into this even it's not a fxpak issue. Looking forward to the patch.

mrehkopf, 2022-02-01:
As a quick patch if you have a hex editor you could try the following - load the ROM and, at offset 0x65-0x66, change "10 00" to "00 01".
Basically the correct instruction should read "stz $0100" but they put "stz $0010" instead. (The byte order is reversed on the 65c816 so "10 00" means $0010 and "00 01" means $0100.)

mrehkopf, 2022-02-01:
(I lied - it's a two byte patch.)

pseiler, 2022-02-01:
Awesome! I'll try it on Thursday and report if it works for me too!
Thank you again.

pseiler, 2022-02-07:
It worked but not as stable as I've thought. When moving around the cursos in the title menu it freezes after a while and ingame had similiar issues in the pause menu. 
I'll test again today and also play a longer session (trying also to disable ingame hooks) to see if I can reproduce the errors.

But thank you very much for your effort!
I'll leave it open till I have further news. I will also buy the release from Strictly Limited Games to see if those issue got fixed and check the differences.

pseiler, 2022-05-11:
I think this can be closed. I also modded another SNES to have a true NTSC master clock / DFO which fixes all the issues I had.
But the patch works so far that you can still play the game on PAL consoles with just the 50/60 Hz Switch/Mod with a few hiccups or freezes.

closed 2022-05-11 by pseiler