mrehkopf/sd2snes #173: Save States not working at all on FXPAK Pro

zkat, 2022-01-11:
Using the latest beta (1.11.0b1) on a brand new FXPAK Pro + SuperNT, save states don't seem to work at all. I've double-checked that in-game hooks are disabled, and this happens whether I have save slots enabled or not.

I've tried multiple games, including:

* Donkey Kong Country
* Aladdin
* Final Fantasy III

Is there any way I can help debug this further, or any other clues as to what might be going on? I don't know any assembly stuff but I'm happy to help QA this one.

mrehkopf, 2022-01-11:
In-game hooks must be enabled for the savestates to work. The built-in save state handler is now called from the in-game hook as opposed to the USB hook to free it up for whatever USB features may be used.

zkat, 2022-01-11:
Enabling in-game hooks does not seem to work either, unfortunately.

zkat, 2022-01-11:
Update: After putzing around a bit, I found that if I mash the button combinations a bunch, they eventually stick. So pressing Start+R a bunch of times eventually makes my screen flash with a save state :)

I guess there's a timing issue? Or maybe my (bluetooth) controller is causing problems?

DoctorDan1986, 2022-01-11:
Have you tried pressing and holding the button Combo?  That might help narrow down a timing issue.

zkat, 2022-01-11:
pretty sure this is just an issue with my controller, so I'm going to close this. I'll reopen if I have reason to believe it's not just a controller timing thing.

closed 2022-01-11 by zkat