mrehkopf/sd2snes #174: i want to compile this code

gizaha, 2022-04-09:
Hello. I clone this code in visual studio and i'm serarching a way to compile it and make some tweaks/experiments.

I can't run it in VS because the run button says "select startup item".
Can anyone help me? It will gain time.
Thanks.

Skarsnik, 2022-05-20:
Information on how to build the firmware are in https://github.com/mrehkopf/sd2snes/blob/develop/src/README

mrehkopf, 2022-05-20:
The code isn't organized as a Visual Studio project (it might be possible but it uses other tools instead).
I think the easiest way would be to download arm-none-eabi-gcc (C compiler for ARM - depending on your OS this may be available as a ready-made package) and build from the command line using make inside the `src` folder, e.g. `make CONFIG=config-mk2`
A shell with assorted tools is definitely required. I use msys2-mingw64 on Windows (https://www.msys2.org/).
An FPGA configuration file is baked into the firmware image at some point, so it will likely complain that it doesn't know how to build `../verilog/sd2snes_mini/fpga_mini.bit` or similar. This file is not mandatory for firmware operation, just for convenience so it can display some error message in case the SD card is unusable. For this purpose you can just replace it with some small text file, just so the make process can find it.

gizaha, 2022-05-21:
Thank you very much for you answer. Howerver i'm too old to dive into this.
Thanks again for your time.

gizaha, 2022-07-28:
> https://github.com/alttpo/sd2snes-build-docker
> 
> I made a docker image that can compile the firmware code and the verilog code. Hope it helps someone.

Any easy guide of how to use docker in visual studio? I have this project opened in visual studio, i want to edit it, compile and test it in sd2snes. 
I'm no programmer, i'm having a hard time with visual studio and I learn of docker today.

JamesDunne, 2022-07-28:
Assuming by Visual Studio you mean Visual Studio Code, I can help. First, let's skip the IDE and go straight to a WSL linux shell and compile the code with the repo I linked above.

Start by installing WSL (windows subsystem for linux) on your Windows machine and then set up Docker `sudo apt install docker.io` within WSL. Google around for guides on how to do all that.

From there, `git clone` the sd2snes-build-docker repo I linked earlier and run the build script. It should clone the sd2snes repository to a working copy and then start building a docker container which contains all the compilers and tools needed to build the code. It should take about 10-20 minutes to complete the first time and then be cached for subsequent builds. The time is mostly dominated by the fpga / verilog toolset download and installation. From there it will copy in the sd2snes working copy code and compile most of it (excluding the extra FPGA bi3 files). The final result should be a couple of .bi3 files that you can copy to your sd2snes SD card's `/sd2snes/` folder. Details in the README.

VS Code and IDE usage is outside the scope of the sd2snes-build-docker repo I put together, but it should be easily accomplished:
1. install VS Code
2. install the "Remote - WSL" extension (Id: `ms-vscode-remote.remote-wsl`) - this extension allows VS Code running on Windows to work on files that live in WSL (linux).
3. open the sd2snes repo working copy
4. edit code as desired
5. change the `src/VERSION` file to make sure the firmware code gets updated
6. run the build.sh script to recompile the .bi3 files

HTH

gizaha, 2022-07-28:
No, i mean visual studio (2017).

JamesDunne, 2022-07-28:
I'm sure you can create your own C/C++ solution for VS2017 containing the firmware C code and edit the code that way to get nice features like auto-completion but beyond that VS2017 won't help you with this project since it needs to be compiled for an ARM Cortex M7 microcontroller. As far as I know, VS2017 C/C++ primarily targets Windows environments and ABIs; you probably won't find much support for the `arm-none-eabi` triplet in VS2017.

Anyway, I suggest you abandon hope for VS2017 and instead try what I posted above to at least be able to compile the firmware and verilog code and get it running on the sd2snes device. I know it works otherwise I would not have created and shared the repo 😄. From there you can worry about your editor of choice.

RandalLinden, 2022-08-01:
@JamesDunne Thank you for the repo -- I found it helpful!

On Windows 11 Ubuntu I found that it requires starting a second shell and doing a "sudo dockerd" to get the daemon running.

Then doing a "sudo ./build.sh" works as expected.

Unfortunately, I'm not familiar with Docker so I would really appreciate a quick example of extending the Docker file to support compilation of the GSU firmware!

Thanks in advance!