mrehkopf/sd2snes #175: FX Pak Pro Savestates not working with Shadowrun ROM.

pecodeguile, 2022-05-20:
The game seems to save the state just fine, but then freezes when I try to load the state.  The game utilizes all of the buttons, so with Start, Select, L, or R you are bringing up some sort of menu engagement.  

I also tried this game with savestate2snes and that didn't work either.  

If there's any more info I can provide, I'd be more than happy to!

pecodeguile, 2022-06-23:
Does this mean that with the latest update it will work? or still a work in progress?  Either way, thank you for your effort!

mrehkopf, 2023-01-30:
Can you try with v1.11.0? I fixed a general issue with freezing when loading a state. If it doesn't, there's a chance it can be fixed with an entry in the savestate_fixes.yml file that I can figure out for you.

pecodeguile, 2023-01-30:
Hello!  And thank you for your work on this!

In Shadowrun, the START button brings up a menu, while the SELECT button
brings up your stats. With the original savestate button mapping (START +
R), the game would freeze when trying to exit the menu after loading a
savestate (START + L).

I altered the config.yml file so that saving a state would be SELECT + R
and loading would be SELECT + L.  This seemed to work at first! I would
save a state, and then load a state, exit out of the status menu and all
looked well!  However, the game then froze whenever combat was initiated.
I could run around in safe areas for seemingly however long I wanted to,
but the second that I got near an enemy, the game would freeze.

Hope this helps!  Thank you again for your effort, and please let me know
if I can be of any further assistance!

-peco

redacted173, 2023-02-04:
Try adding the following line to sd2snes/savestate_fixes.yml on the SD card:

F834: 001D00,2140 # Shadowrun (US)

pecodeguile, 2023-02-06:
We are definitely getting closer!  This time the game went into combat just
fine!  In fact, I played for 10 minutes and was still able to load an early
and late savestate.  However, when I used L+R+Select+Start to reset the
game and then load a new game, it would no longer load the previous
savestate.

It would show the loaded savestate for a brief second, then go completely
black screen and unresponsive.  It was unplayable after that.

Thank you for your continued effort on this!

redacted173, 2023-02-06:
I'm not able to reproduce it.  I suggest:
- disabling save state slots in the menu.  I'm not sure if an empty slot loads uninitialized memory or something useful.
- mapping save,load to at least one additional button from the reset/restart in-game buttons.  It's unfortunate that there is overlap.  For example B+R,B+L

Try using it more and seeing where it fails and where it doesn't consistently.  Ikari may have other ideas since he knows the implementation better.

mrehkopf, 2023-02-07:
About overlap: Is it really necessary to allow the non-exclusive button combinations (i.e. allow other buttons to be pressed) for the save state features? It compromises a lot of the other button commands...

mrehkopf, 2023-02-07:
And yes, the handler will still attempt to load an uninitialized save state, so that can be either just zeros or an existing save state of a game played earlier. Both will lead to undesired behavior :) But in this case I think it could be the overlap - since the "reset game" combination also contains the "save state" combination it might attempt saving a state while the microcontroller is already in the process of resetting the SNES.

redacted173, 2023-02-07:
Complete overlap may be functional.  The suggestions are trying to eliminate some possible races with button presses.

It certainly could lead to an undesirable situation where the state is saved over when someone tries to reset.  It all depends on how strict the button checking is done and the order the buttons are pressed.  I typically disable in-game buttons and reset using the console buttons.

redacted173, 2023-02-07:
Would it possible to only perform one check per interrupt and have a jump table or self modifying code to change what is checked each time?  Seems like that could help eliminate the complexity of the offload and also reduce overhead.  Not sure it's necessary to check all conditions on every interrupt.

mrehkopf, 2023-02-07:
Sure but I intend to to it using the FPGA instead of 65816 code to save cycles. The current in-game hook implementation (except the save state handler, which is my TODO) uses the FPGA to make decisions about branching and inject branch offsets. Doing everything in 65816 consumed too many cycles which led to compatibility problems with some games.

pecodeguile, 2023-02-08:
Hello!

I remapped the savestate button commands to BR and BL and it works!  I was
able to turn off the console, remove the fxpak, then start the game over
and load a savestate directly where I had last saved it!

Great job!  Well done!  And thank you ever so much for all of your work on
this!

lauuurent, 2025-03-14:
Hello !

### I encountered the same problem on Donkey Kong Country: the game freezes when loading a game, probably because of a conflict with the "START" button.

> - Can you detail the procedure to follow to correct this problem ?
> (What code to enter, in which file, and if there are other operations to do?)
> 
> - Furthermore, I hesitate to use the combination B+L or B+R, because some games require it. Do you think SELECT would be more suitable ?
> If not, what other combination would you recommend ?

Thanks in advance !