mrehkopf/sd2snes #177: Cheats and SA-1

Ritaku02, 2022-05-30:
Can the cheat system access SA-1 memory space? I converted some Gradius III cheats to work on the SA-1 patched version, and they work fine in RetroArch with Snes9x core, but not on the SD2SNES. It leaves me wondering if it's the cheats system being unable to access SA-1 memory space, or if it's being mapped differently from RetroArch/Snes9x.

Stingray77, 2022-06-27:
I think the cheat engine works differently. Can you apply the cheats to the Rom itself? Using GameGenieGuy for instance? maybe this also fails as the addressing is changed. but worth a try. 😄

Ritaku02, 2022-06-29:
That would depend on if a cheat addressing the SA-1's RAM can be converted to a Game Genie code, which I'm doubtful of. Game Genie tends to only work on ROM space

Stingray77, 2022-06-29:
Maybe patch the rom before doing the SA-1 patch? (worth a try)