mrehkopf/sd2snes #178: SGB audio garbled with Kirby's Dreamland 2?

jakobdavid44, 2022-06-10:
Just started using my FXPak Pro, so far everything is either awesome or just still needing formatting on my part. Still, some great stuff so far!

The only major issue so far I'm noticing is that when I play Kirby's Dreamland 2, a one or two sound channels sound garbled when certain music and sound effects play.

Is there something I need to do, perhaps a bad ROM? Or is this something that affects other games, and needs to be fixed emulation-side?

jakobdavid44, 2022-12-04:
Hi, thanks for your reply.

To answer your question, I've only gotten through the first world so far and the sound channel garbling already can be heard in the first few levels. Keeping in mind, this doesn't affect the level-select screen or the title screen, only the levels during gameplay proper. The best way I can describe it is that in the first world's levels, the main theme almost sounds like the audio has been compressed, so it sounds more crackly in general compared to proper audio.

It's not my best analogy, but I hope it's a good start!

redacted173, 2022-12-05:
You can try the files in the attached zip.  Not sure what else it could be since I'm having a difficult time hearing the problem.

[kdl2_nrx2and3_no_update_duty_pos.zip](https://github.com/mrehkopf/sd2snes/files/10149866/kdl2_nrx2and3_no_update_duty_pos.zip)