mrehkopf/sd2snes #183: Bad MSU-1 and SGB audio channel balance on SNES Jr

seankhl, 2023-01-05:
I've tested this on two Snes Jrs and had the same result on both: the right channel's audio is much quieter than the left's (maybe by about 8db). Was super frustrating to find that my second console had the same issue as the first, and I began wondering if it's really an issue with the console at all.

But, I tested on my friend's Analogue Super NT and did not have the issue. Maybe I got extremely bad luck with my SNES Jrs, or maybe it's something to do with the SNES Jr revision itself?

Would be happy to provide any other info as necessary to help solve this. Even if it ends up being a strange idiosyncrasy with SNES Jr only, having an option to adjust the on-cart audio channel balance would be greatly appreciated. Obviously I can't just balance the audio output myself because it will affect the console audio balance.

mrehkopf, 2023-01-05:
I have multiple SNES Jr. and cannot confirm this issue at least for my units. Which sd2snes/FXPAK hardware revision are you using?

My current guess is this: The SNES has pretty low impediance on the Line input pins on the cart slot - 200 Ohms per channel. If for some reason one of the audio outputs on the FXPAK has become high impedance (caused by, e.g. a faulty ferrite bead on one of the channels, or some passive surrounding the opamp, or the opamp itself), the load resistor in the SNES on that channel will reduce the voltage of the audio signal more than on the other channel.

The Super Nt, not having low-impedance line inputs - I just measured around 4 Megaohms, which is parasitic, so basically open line - would not affect the signal in the same way, therefore the difference would be less noticeable, if at all.

seankhl, 2023-01-05:
Hi, it's an FXPAK Pro bough about 2 years ago.

That seems like a good guess. I have some notes from the last time I was debugging this (a couple months after I bought it) and I did test another sd2snes which didn't have the issue. So, I think it may be sensible to pin it down to the cart, and your explanation sounds like a great match for what I'm seeing. Is there anything actionable I can do to confirm or fix this? I know this is a issue board for the firmware, and I posted here with the thought that it could be firmware-related, but regardless any advice would be greatly appreciated.

mrehkopf, 2023-01-06:
If you have a multimeter you could do some measurements. I've attached a picture but I don't know if it's readable very well ^^;
On the measurements between the chip and C391 (the fat capacitor) the line colors denote the probe polarity (i.e., negative probe to the IC pin, positive probe to the end of the capacitor). Polarity doesn't matter for the other measurements.
On the two coupling capacitors (C351, C352) the resistance will be low initially (probably measures around 50kOhms), but will rise increasingly as the capacitors charge. If the resistance stays low (like, below a few kOhms) then that capacitor is bad.

[image]

closed 2023-06-25 by mrehkopf