mrehkopf/sd2snes #184: 1.10.3 OK 1.11.0 KO

josete2k, 2023-01-28:
After the Near's splash screen the sd2snes freezes or have graphical glitches and freezes...

Back to 1.10.3 and it's all ok again.

1CHIP PAL supercic snes (LM78S05 regulator too) 

REVH modded board.

josete2k, 2023-01-28:
[image]

1.11.0b1 same issue

josete2k, 2023-01-28:
[image]

1.10.3 working

mrehkopf, 2023-01-29:
Wild. Can you run the diagnostic firmware (https://sd2snes.de/files/sd2snes_diagnostic_firmware.zip) and attach the entire log? You can find instructions for running the firmware at https://sd2snes.de/blog/cool-stuff/sd2snes-diagnostics
The new firmware depends more on a specific clock signal from the SNES and I want to check if it arrives at the FPGA properly. 

NOTE to anyone else reading this issue: the diagnostics I linked is only for the OG sd2snes.

josete2k, 2023-01-29:
Here you have:

https://pastebin.com/xqjuH3rK

`===log opened===
derp
SD test... please insert card
=============================
data=111133b04504
crc1 = d805a1bf crc2 = d805a1bf   PASSED

RTC Test
========
setting clock to 2011-01-01 00:00:00
2011-01-01 00:00:00
waiting 5 seconds
new time: 2011-01-01 00:00:04
PASSED

FPGA test
=========
configuring fpga...
fpga_pgm: 212501 bytes programmed
FPGA configured
basic communication test...PASSED

RAM test
========
Testing RAM0 (128Mbit) - clearing RAM - writing RAM - verifying RAM -RAM0 PASSED

Testing RAM1 (4Mbit) - writing RAM - verifying RAM...RAM1 PASSED

SD DMA test
===========
clearing RAM - loading test file - verifying -PASSED

Loading SNES test ROM
=====================

sysclk test
===========
measuring SNES clock...
21475787 Hz 
21475788 Hz diff = 1  max = 1
21475788 Hz diff = 0  max = 1
21475788 Hz diff = 0  max = 1
   CPUCLK: 3420045
  READCLK: 3094326
 WRITECLK: 162859
  PARDCLK: 162859
  PAWRCLK: 162859
  REFRCLK: 78725
ROMSELCLK: 488578
PASSED

SNES test status: IRQ: 00   PA: 00   MEM: 00/00
SNES test status: IRQ: 5a   PA: 5a   MEM: 00/00
SNES test status: IRQ: 5a   PA: 5a   MEM: 00/01
SNES test status: IRQ: 5a   PA: 5a   MEM: 00/02
SNES test status: IRQ: 5a   PA: 5a   MEM: 00/03
SNES test status: IRQ: 5a   PA: 5a   MEM: 00/04
SNES test status: IRQ: 5a   PA: 5a   MEM: 00/05
SNES test status: IRQ: 5a   PA: 5a   MEM: 00/06
SNES test status: IRQ: 5a   PA: 5a   MEM: 00/07
SNES test status: IRQ: 5a   PA: 5a   MEM: 00/08
SNES test status: IRQ: 5a   PA: 5a   MEM: 00/09
SNES test status: IRQ: 5a   PA: 5a   MEM: 00/0a
SNES test status: IRQ: 5a   PA: 5a   MEM: 00/0b
SNES test status: IRQ: 5a   PA: 5a   MEM: 00/0c
SNES test status: IRQ: 5a   PA: 5a   MEM: 00/0d
[... 422 lines cut ...]
SNES test status: IRQ: 5a   PA: 5a   MEM: 01/b9
SNES test status: IRQ: 5a   PA: 5a   MEM: 01/ba
SNES test status: IRQ: 5a   PA: 5a   MEM: 01/bb
SNES test status: IRQ: 5a   PA: 5a   MEM: 01/bc
SNES test status: IRQ: 5a   PA: 5a   MEM: 01/bd
SNES test status: IRQ: 5a   PA: 5a   MEM: 01/be
SNES test status: IRQ: 5a   PA: 5a   MEM: 01/bf
SNES test status: IRQ: 5a   PA: 5a   MEM: 01/c0
SNES test status: IRQ: 5a   PA: 5a   MEM: 01/c1
SNES test status: IRQ: 5a   PA: 5a   MEM: 01/c2
SNES test status: IRQ: 5a   PA: 5a   MEM: 01/c3
SNES test status: IRQ: 5a   PA: 5a   MEM: 01/c4
SNES test status: IRQ: 5a   PA: 5a   MEM: 01/c5
SNES test status: IRQ: 5a   PA: 5a   MEM: 01/c6
SNES test status: IRQ: 5a   PA: 5a   MEM: 01/c7
SNES test status: IRQ: 5a   PA: 5a   MEM: 01/c8
SNES test status: IRQ: 5a   PA: 5a   MEM: 01/c9
SNES test status: IRQ: 5a   PA: 5a   MEM: 01/ca
SNES test status: IRQ: 5a   PA: 5a   MEM: 01/cb
SNES test status: IRQ: 5a   PA: 5a   MEM: 01/cc
SNES test status: IRQ: 5a   PA: 5a   MEM: 01/cd
SNES test status: IRQ: 5a   PA: 5a   MEM: 01/ce
SNES test status: IRQ: 5a   PA: 5a   MEM: 01/cf
SNES test status: IRQ: 5a   PA: 5a   MEM: 01/d0
`

shadowofdarkness, 2023-01-29:
A year ago I had the same problem where the 1.11.0 beta would start up corrupted or just a black screen but I could run 1.10.3 and play basic non extra chip games. Turns out that my caps where just starting to go bad and I was told the 1.11.0 version actually used more power even just at the menu and that is why it had more problems.
A full capacitor replacement fixed all my problems.

You could have the same issue. Assuming you just upgraded the regulator and left the original caps in.

josete2k, 2023-01-29:
I can play every extra chips (yoshi's Island MSU too) in 10.3

I only replaced the regulator. But, do you Really think that it is a capacitor issue? I'll take a look if this issue cannot be solved by software...

Thanks.

mrehkopf, 2023-01-29:
It could be related to the whole save state logic running along, now triggering some underlying defect; unfortunately your diagnostic log is incomplete, it is missing part of the SNES memory test and the test summary. Did you wait until the LEDs made a running light and the screen turned (hopefully) green?
But the clocks are listed in the log, I think they are a bit high but I need to confirm with reference values.
It is a Brazilian SNES, correct?

mrehkopf, 2023-01-29:
Oh, was just guessing from the unusual system clock (21.45MHz).

josete2k, 2023-01-29:
Well... That's maybe because the supercic mod (with 17.734476Mhz Crystal and a 21.47727Mhz Crystal), isn't it?

josete2k, 2023-01-29:
Full log

https://pastebin.com/xY7JmDLn

===log opened===
derp
SD test... please insert card
=============================
data=111133b04504
crc1 = d805a1bf crc2 = d805a1bf   PASSED

RTC Test
========
setting clock to 2011-01-01 00:00:00
2011-01-01 00:00:00
waiting 5 seconds
new time: 2011-01-01 00:00:04
PASSED

FPGA test
=========
configuring fpga...
fpga_pgm: 212501 bytes programmed
FPGA configured
basic communication test...PASSED

RAM test
========
Testing RAM0 (128Mbit) - clearing RAM - writing RAM - verifying RAM -RAM0 PASSED

Testing RAM1 (4Mbit) - writing RAM - verifying RAM...RAM1 PASSED

SD DMA test
===========
clearing RAM - loading test file - verifying -PASSED

Loading SNES test ROM
=====================

sysclk test
===========
measuring SNES clock...
21475788 Hz 
21475787 Hz diff = 1  max = 1
21475788 Hz diff = 1  max = 1
21475787 Hz diff = 1  max = 1
   CPUCLK: 3420044
  READCLK: 3094326
 WRITECLK: 162859
  PARDCLK: 162859
  PAWRCLK: 162859
  REFRCLK: 78725
ROMSELCLK: 488578
PASSED

SNES test status: IRQ: 00   PA: 00   MEM: 00/00
...
SNES test status: IRQ: 5a   PA: 5a   MEM: 5a/00

TEST SUMMARY
============

flash check:   1111-4-33b04504-dfb09a7f-08-11-0  
Test      Result
----------------
SD        Passed
USB       not implemented
RTC       Passed
CIC       no run
FPGA      Passed
RAM       Passed
SD DMA    Passed
CLK       Passed
DAC       not implemented
SNES IRQ  Passed
SNES RAM  Passed
SNES PA   Passed

mrehkopf, 2023-02-06:
Can you check if the v1.10.3-frs firmware https://github.com/furious/sd2snes/releases/tag/savestate_v12
Or the https://github.com/RedGuyyyy/sd2snes/releases/tag/usb2snes_v11 works?
Both need to be installed over an existing v1.10.3.

josete2k, 2023-02-06:
[image]
[image]

Yes

mrehkopf, 2023-03-12:
Could you try this FPGA bit file? Just overwrite your original fpga_base.bit in your sd2snes folder with the one from the ZIP file. (Only for OG sd2snes and no expansion chips for now)
[fpga_base_new_address_sampling.zip](https://github.com/mrehkopf/sd2snes/files/10951983/fpga_base_new_address_sampling.zip)

josete2k, 2023-03-12:
I'll test it tomorrow.

Thanks in advance.

josete2k, 2023-03-26:
> Could you try this FPGA bit file? Just overwrite your original fpga_base.bit in your sd2snes folder with the one from the ZIP file. (Only for OG sd2snes and no expansion chips for now) Firmware v1.11.0 needs to be installed first. [fpga_base_new_address_sampling.zip](https://github.com/mrehkopf/sd2snes/files/10951983/fpga_base_new_address_sampling.zip)

Nothing. Black screen.

mrehkopf, 2023-09-17:
https://sd2snes.de/files/sd2snes_v1.11.0-testsampling.zip
Can you extract this over your current v1.11.0 installation (so that fpga_base.bi3 and fpga_base.bit are overwritten) and try if it works better?

josete2k, 2023-10-21:
> https://sd2snes.de/files/sd2snes_v1.11.0-testsampling.zip Can you extract this over your current v1.11.0 installation (so that fpga_base.bi3 and fpga_base.bit are overwritten) and try if it works better? This basically brings back the old sampling as it was in 1.10.3.

Sorry for the later reply

It works!!! But some games like Batman returns PAL show the region warning: "this game is not designed for your console"

Same rom is working with 1.10.3

SuperCIC enhanced ON
menu 60Hz 
game auto  

I've tested with menu 50 and game 50 and same issue... the console remains at 60Hz in game even menu is 50hz

mrehkopf, 2023-10-21:
Oh, try this fix: https://github.com/mrehkopf/sd2snes/issues/186#issuecomment-1434480057

(Auto region patching doesn't seem to work well in some cases; I suspect that wrongly dimensioned parts are used on some old cartridges. This might also be causing the original problem that you reported in this issue. Could you post some pictures of your cart?)

josete2k, 2023-10-22:
> Oh, try this fix: [#186 (comment)](https://github.com/mrehkopf/sd2snes/issues/186#issuecomment-1434480057)
> 
> (Auto region patching doesn't seem to work well in some cases; I suspect that wrongly dimensioned parts are used on some old cartridges. This might also be causing the original problem that you reported in this issue. Could you post some pictures of your cart?)

Working...

I've mixed https://sd2snes.de/files/sd2snes_v1.11.0-testsampling.zip and https://github.com/mrehkopf/sd2snes/files/10765998/issue186.zip and it seems that's all ok.

josete2k, 2023-10-22:
[url=https://postimg.cc/0zgztgbb][img]https://i.postimg.cc/0zgztgbb/IMG-20231022-180504.jpg[/img][/url]

[url=https://postimg.cc/qz6tXBBP][img]https://i.postimg.cc/qz6tXBBP/IMG-20231022-180516.jpg[/img][/url]

It's a chinese cart with the mod to rev H

josete2k, 2023-10-22:
[image]
[image]

josete2k, 2023-10-22:
This is the pcb that I used with magno for sdd1 tests and there were some issues with sram files on it (issues you fixed in the past)...

r0g1nh0, 2024-08-30:
Thank you very much josete2k, my sd2snes rev F was not saving the games or playing the audio of the MSU-1 games and with the combination below, everything started working again.

I've mixed https://sd2snes.de/files/sd2snes_v1.11.0-testsampling.zip and https://github.com/mrehkopf/sd2snes/files/10765998/issue186.zip and it seems that's all ok.

mrehkopf, 2026-05-16:
How's v1.11.1 working out for you?

josete2k, 2026-05-16:
Just see now the 1.11.1

I'll test in a few days!

Thanks!!!