mrehkopf/sd2snes #185: D4 patch doesn't work with v1.11.0

Romain80200, 2023-01-29:
I’ve got a bug with d4 patch with this release but it work with the beta.
D4 patch doesn’t work with this release.

mrehkopf, 2023-01-29:
FXPAK and SuperCIC don't play well together if SuperCIC enhanced mode is disabled and other SuperCIC related mods are built into the console.
The reason is that the console SuperCIC can never know the correct region of the game; it guesses it based on the cartridge CIC's region. This only works well with original games. On the sd2snes/FXPAK the cartridge CIC can change region based on the console it is inserted to. Also, the region that the cartridge CIC signals to the console cannot change per game. Probably you got lucky before since your cartridge's CIC was set to PAL but due to some contact issue is now switched to NTSC. Therefore now the SuperCIC in your console will always think that the game is NTSC, and the D4 mod itself will patch to NTSC even though the game expects PAL.

Bottom line: The only sane way to run a SuperCIC cartridge combined with a SuperCIC console is to enable SuperCIC enhanced mode in the menu; you will then have the possibility to change the screen mode independent of the game's region, and it will always signal the correct region to your D4 mod as well.

mistepien, 2023-02-18:
To cut the chase: if SuperCIC mode is turn on in FXPAK menu then should we turn off D4 patch in SuperCIC hardware mod?

mrehkopf, 2023-02-18:
Kinda vice versa - if a D4 mod is installed in the SNES then the SuperCIC mode should be enabled in the FXPAK menu.
If you do not wish to use the SuperCIC mode then the D4 patch of the SuperCIC mod should be disabled.

Romain80200, 2023-02-26:
Hello,
With the [issue186-try2.zip](https://github.com/mrehkopf/sd2snes/files/10778271/issue186-try2.zip) firmware patch, everything is working ok, region patch, video mode, cic ehanced mode, even the led are working again.

closed 2023-03-12 by mrehkopf