mrehkopf/sd2snes #186: [v1.11.0] Dead LEDs and missing card error message

Ramsis-SNES, 2023-01-29:
First of all, thanks for the new firmware release, ikari. :+1: 

Two small (?) issues:

- The green and yellow LEDs are dead. Upon power-up (with an SD card inserted), the green LED turns on very briefly, never to light up again afterwards. The yellow LED never shows any sign of life at all. Without an SD card inserted, the green LED lights up, and the red LED flashes. That one (the red LED) at least seems to operate normally.
- With no card inserted, the "No SD Card found!" message won't show up anymore (gray screen).

Tested on both Super Nt and a SuperCIC-enhanced PAL SNES, the results are the same in both cases. I even deleted my config.yml file to rule out any potential issues with that, to no avail. The LED brightness setting doesn't make a difference either, apart from affecting the red LED, obviously.

mrehkopf, 2023-01-29:
Oh, and here I thought my old weathered MK.II's LEDs were just broken... :sweat_smile:

mrehkopf, 2023-02-17:
Made some fixes in #196 - this also contains another fix which addresses an issue where the wrong video mode is set in SuperCIC pair mode. Please give it a try :) 
(Only firmware flash images are included - copy on top of an existing v1.11.0 firmware)
Download: [issue186.zip](https://github.com/mrehkopf/sd2snes/files/10765998/issue186.zip)

mrehkopf, 2023-02-17:
Oh, forgot to mention - I can't reproduce the boot error message problem, always shows for me even on Super Nt :/

Ramsis-SNES, 2023-02-17:
> Made some fixes in #196 - this also contains another fix which addresses an issue where the wrong video mode is set in SuperCIC pair mode. Please give it a try :)

Awesome! I just tested it and the LEDs do indeed light up again. Thanks! :-)

> Oh, forgot to mention - I can't reproduce the boot error message problem, always shows for me even on Super Nt :/

Hmm, weird. It's no big deal of course, but it kind of bothers me as it used to work just fine ... I'm going to try it on at least one other console with different chip revisions to see if that makes a difference.

mistepien, 2023-02-18:
> Made some fixes in #196 - this also contains another fix which addresses an issue where the wrong video mode is set in SuperCIC pair mode. Please give it a try :) (Only firmware flash images are [...]

It changes nothing with my SuperCIC-only-NTSC issue.

mrehkopf, 2023-02-19:
I cannot find any mention of the issue by you before - what exactly is the issue? Maybe reference or open a new one. Thanks.

mistepien, 2023-02-19:
Sorry, I thought that by " wrong video mode is set in SuperCIC pair mode" you mean https://krikzz.com/forum/index.php?topic=12376.msg88164#msg88164

mrehkopf, 2023-02-19:
Ah, yes, that's the one. So pair mode is enabled (you can change the menu video mode using the SuperCIC settings menu) but a game always runs at 60Hz despite the setting?
(Two SuperCICs (both in console and on cart) aren't really deterministic regarding the detected region and should not be used outside of pair mode, especially when a D4 mod is also involved inside the console)

mistepien, 2023-02-19:
Exactly that happens. How can I turn off one SuperCIC? I have FFVIMAN mod on 1CHIP PAL SNES.

mrehkopf, 2023-02-19:
If you do not wish to use the pair mode but control video mode using the reset button, you can disable SuperCIC in the SuperCIC settings (needs power cycle). However if your SNES SuperCIC mod has a D4 mod integrated it will try patching to constant region reported by the SuperCIC which may not reflect the game being played. The pair mode is designed to circumvent this limitation.
Which sd2snes or fxpak are you using, and did you confirm that the firmware linked above was actually installed? (Check System Information screen, it should say "Firmware version: issue186".

mrehkopf, 2023-02-19:
Found a problem with the Mk.3 (FXPAK Pro) firmware build, the header magic was wrong due to a copy&paste error (see 13a4af4) so it wouldn't update itself... 
Please try again with this if you have an FXPAK Pro: 
[issue186-try2.zip](https://github.com/mrehkopf/sd2snes/files/10778271/issue186-try2.zip)

mistepien, 2023-02-19:
Great! Actually I was wondering why the version was still 1.11 even if I copied files. I will check out morning. Thanks!

mrehkopf, 2023-02-20:
Yes, the Super Nt has a modified SuperCIC built in. But sadly it does nothing with it, there's no way to do region switching, it is simply ignored by the firmware.

mistepien, 2023-02-20:
>Please try again with this if you have an FXPAK Pro: [issue186-try2.zip](https://github.com/mrehkopf/sd2snes/files/10778271/issue186-try2.zip)

It works like a charm. FXPAK PRO switches NTSC/PAL as it should.

mrehkopf, 2023-02-20:
Ah, I'm relieved. Closing this issue then, next firmware release will have the fix built in :)

closed 2023-02-20 by mrehkopf

mistepien, 2023-02-20:
BTW, did you encounter situation like that with SNES: sometimes FXPAK PRO does not want to boot (green led is blinking) or sometimes gamepad 1 does not work while gamepad 2. If it happens after turning off and turning on FXPAK does not boot. Is it a case of so-called Freak SNES (https://sd2snes.de/blog/faq)?

mrehkopf, 2023-02-20:
Green LED blinking usually means that no firmware was found inside flash. Should never occur when a firmware has been used on the FXPAK at any point in time :) It could be a power supply related issue, the controller thing too.

mistepien, 2023-02-20:
Thank you very much! You and Krikzz are doing great job! FXPAK PRO + SuperCIC + MSU-1 is just mind blowing.

Ramsis-SNES, 2023-02-20:
> Oh, forgot to mention - I can't reproduce the boot error message problem, always shows for me even on Super Nt :/

I just tested the cart with no SD card inserted (and with firmware "issue186" as well as the latest menu.bin with the favorite games fix installed) on three different consoles. Surprisingly, the behavior seems to have changed when compared to firmware v1.11.0 (2-PPU SNES):

- Super Nt: Card error message won't show up (solid gray screen, same as with official v1.11.0); pressing the Reset button makes no difference, and the "Show video mode" option in Debug settings doesn't do anything at all (if that helps/matters)
- 1CHIP Ultra16: Card error message appears, but only in Autoboot mode (skipping U16 menu) (not tested with official v1.11.0); running the cartridge from the menu results in a solid gray screen
- 2-PPU SuperCIC-modded PAL SNES: Card error message appears (as opposed to official v1.11.0, where it wouldn't show up on this exact same console), but only upon cold boot; pressing the Reset button results in a solid gray screen.

@mrehkopf On a regular SNES, does the error message still appear after pressing Reset? (Super Nt might be a different story altogether due to custom system settings, so let's just ignore that for now ... if you'd like to pursue this at all.)

Thanks!

mrehkopf, 2023-02-20:
Resetting never shows the text again since the ROM cleans up the text buffer for the MCU to fill in.
When you run the cart from the Ultra16 menu the MCU will already have filled in the text at that point, only for the ROM to clear it again. ^^

mrehkopf, 2023-02-20:
Actually that would also explain the behaviour on Super Nt. I wonder why it's been working for me 🤔

mrehkopf, 2023-02-21:
Couldn't resist and opened #201. xD Revamping the display a bit.

Ramsis-SNES, 2023-02-21:
> Resetting never shows the text again since the ROM cleans up the text buffer for the MCU to fill in. When you run the cart from the Ultra16 menu the MCU will already have filled in the text at that point, only for the ROM to clear it again. ^^

Ah, I see, that makes sense. Thanks for explaining!