mrehkopf/sd2snes #189: Save State issue - Dragon Quest 5 Translation Patch

skins4thewin, 2023-02-02:
After loading a saved state, the sound effects stop & the game freezes after you take pretty much any action.  Aka making a menu choice, talking to someone, changing rooms, etc.  This has been an ongoing issue with this game since Save States were first introduced.

redacted173, 2023-02-10:
Try adding the following to sd2snes/savestate_fixes.yml on the SD card.  The sound engine has a lot of state controlled by multiple functions which make this complicated.  The fix attempts to reset some of this state so it doesn't lock up.

You'll have to test it out to see if it works for you.  If it locks up in a dialog, try saving again to see if it unlocks itself.  If it doesn't work at all you may have a different checksum in your ROM that needs to replace the "E3DA".

E3DA: # Dragon Quest V (T+Eng 2.01F)
  - 002140,00
  - 0008D8,00
  - 0008DA,00
  - 0008DC,00
BAF9: # Dragon Quest V (JP)
  - 002140,00
  - 0008D8,00
  - 0008DA,00
  - 0008DC,00

skins4thewin, 2023-02-11:
Thanks for looking into this!  However you may as well be talking a foreign language lol as I really am not sure what any of that means...  what exactly am I adding to where?

Sent from my Verizon, Samsung Galaxy smartphone

redacted173, 2023-02-11:
Let me know if these directions aren't clear:
(1) Insert your SD2SNES's SD memory card into your computer.  Let's assume it shows up as E: drive in windows.

Now you have two choices:
(2a) Download the attached savestate_fixes.zip at the bottom of this comment.  Unzip it.  Copy the savestate_fixes.yml file into the E:\sd2snes\ directory.  Save yes to copying over the existing file.
-or-
(2b) Open the file E:\sd2snes\savestate_fixes.yml in notepad.  It's a text file and you should see text that looks similar to what I put in my previous comment.
(2c) Copy all 10 lines of text starting on the line with "E3DA: ..." in my previous comment to the end of the file and save it.

(3) Insert the SD card back into your SD2SNES and try savestates.

[savestate_fixes.zip](https://github.com/mrehkopf/sd2snes/files/10712108/savestate_fixes.zip)

skins4thewin, 2023-02-11:
Ahh I see, I didn't see the attached file in the last email.  I get it now.  Thanks, I will definitely try this out next time I boot up the game.  Thanks a bunch for giving this issue some attention! :)

Sent from my Verizon, Samsung Galaxy smartphone