mrehkopf/sd2snes #191: [v1.11.0] Final Fantasy 3/6 - Black/corrupt equip screen with or without in-game hooks

jami3rez, 2023-02-04:
When accessing menus the screen is mostly black and is typically missing information. Sometimes random garbage characters are in the place of letters and numbers.

[image]

This happens irrespective of whether in-game hooks are turned on or off.

No other games have exhibited any problems with my SD2SNES cart and my SNES console passed all suggested testing suites I have thrown at it.

This problem persists across all firmware I have tried and multiple different ROMs, both official and those with hacked ‘fixes’.

mrehkopf, 2023-02-04:
Thanks for reporting; this increases the number of reports with this issue to three, so there seems to be some consistency to it. So far I haven't been able to reproduce the issue myself...
Does it happen every time you enter the equip menu, or just sometimes? Is it possible to reproduce (for me) when entering the equip menu even in the intro sequence (when Terra is still named "?????")?
Finally, what type of SNES are you using, and do you have 1CHIP Transient Fixes or Brightness Limit enabled in the menu?

jami3rez, 2023-02-05:
Hi, thanks for the replies.

I am not using a PAL, non-1CHIP console.

These problems occur with every single option in the SD2SNES Config menu disabled, that includes Save States, Brightness Limit etc. They occur every time I access the Item, Skills, Equip and Relic menus without fail. Status, Config and Save display normally.

I have recreated the issue from a new game with Terra named ????? below.

[image]
[image]
[image]

The following are from a save game about an hour in. This is as far as I have played.

[image]
[image]
[image]
[image]
[image]
[image]
[image]

Ramsis-SNES, 2023-02-05:
Possibly a 50/60 Hz issue, this is known even with an original cart.

mrehkopf, 2023-02-05:
Oh dear, you're right - it's been such a long time that I had completely forgotten about this issue. Yes, it's indeed a game "bug" when run at 50Hz on a PAL console.

jami3rez, 2023-02-05:
> Possibly a 50/60 Hz issue, this is known to happen even with an original cart.
> 
> Edit 2: Try this: [FF6 PAL fix](https://www.romhacking.net/hacks/106/)

Fantastic stuff! This does seem to have done the trick as the menus I mentioned do display normally now.

Couple of caveats. 

1. On the intro sequence of the mechs walking through the snow the ground looks like a chess board. Alternating big black squares. Otherwise fine. Could be an issue with the patched ROM I guess, but not a massive issue.
2. With in-game hooks enabled the overworld map flickers constantly. The menus however continue to display fine.

Thanks again for the help.

closed 2023-02-05 by jami3rez

mrehkopf, 2023-02-05:
Hm! I don't know how the patch works but maybe it just "wastes" cycles to mimic the NTSC timing, up to the point that not enough are left for the in-game hook - in that case it might be possible to loosen the timing of the patch itself a little.

Ramsis-SNES, 2023-02-06:
>     1. On the intro sequence of the mechs walking through the snow the ground looks like a chess board. Alternating big black squares. Otherwise fine. Could be an issue with the patched ROM I guess, but not a massive issue.
Definitely neither an issue with the PAL fix patch nor the sd2snes. Be sure to try the patch on a fresh ROM to rule out any hack incompatibilities. ;-)

> Hm! I don't know how the patch works but maybe it just "wastes" cycles to mimic the NTSC timing, up to the point that not enough are left for the in-game hook - in that case it might be possible to loosen the timing of the patch itself a little.

From my memory, all the PAL fix patch does is re-enable one of the BGs in the menu, and have the sound driver wait for the game engine to catch up in the ending sequence so it won't crash with a black screen. I could be mistaken though ...

StrikerMan780, 2023-05-01:
I get this corruption (but worse) in an NTSC FF3/6 ROM. So, I don't think it's a pal-specific issue. I can't seem to remedy it aside from turning off hooks.