mrehkopf/sd2snes #192: Yoshi's Island Glitches with Big Sprites and 3D elements

rafadorock4, 2023-02-07:
This only happens in the Boss Stages, or other states where big enemies and / or big 3D elements appears. This have always happened with my cart with any firmware version. Such thing only happens in Yoshi's Island, other SuperFX games are fine and other special chip games works just fine too. This has been just a visual / cosmetic issue though, the game does not crashes or gives blank screen, it works perfectly and I could even beat the entire game even with visual glitches going on. I inform that with the genuine cartridge of the game, this does not happens, so it is not my console that is faulty or something...

I'll post screenshots soon.

mrehkopf, 2023-02-07:
Are you possibly running a version of a game that is not matched to your console region? Sounds like a typical 50/60Hz issue :)

rafadorock4, 2023-02-07:
Hi man, maybe I possibly figured out the issue.....

I'm Brazilian, and my console is PAL-M which is the unique color / video type in the World (shits of Brazil...) PAL-M is same as PAL from Europe, but with 60Hz from NTSC.

Below is a test with the 3 regions... First with the EU version:
[image]

Then, with the Japanese version (same as the one I have in the genuine cartridge) - no issues:
[image]

Then, I tested with USA version and, it worked :P ....  - no issues...

And then I retested again and again, over and over, and I found the problem... IPS patched ROM for MSU1 is the thing....
[image]

Look at these lines:
[image]

I've fought with this for years and finally I discovered the issue on mine... Thank you for your help. I think we can close this, because IPS patched ROMs are not the issue with the cart. I'll try to realize myself what is going on with the patched ROMs because it shouldn't corrupt like that and cause artefacts on screen.

closed 2023-02-07 by rafadorock4

paulofloresjunior, 2023-02-07:
Hi Rafa, here is Paulo from Brazil also. Did you try another MSU version?