mrehkopf/sd2snes #194: 1.11.0 - Creating save state causing black screen & music continues

viarurla, 2023-02-16:
Using the FXPAK Pro (not the latest rev) I am having no luck actually creating save states for any games without the screen cutting to black while the music continues in the background.

I've wiped everything including the existing config file and tried games such as Super Metroid (JA/US) and DKC1(US) to no avail. I've tried a multitude of turning game hooks/save states on and off. Are there additional steps I should take to ensure it's not just user error on my part? Everything else with the cart appears to operate flawlessly.

mrehkopf, 2023-02-17:
Did you use a m3nu.bin from a different firmware version? Does it work with Super Mario World?
There's a chance that the state cannot be captured correctly (and restored after saving) because of connection issues; may be worth running https://sd2snes.de/files/sd3snes_diagnostic_firmware.zip as a firmware and checking sd2snes\test_log.txt after completion. (The test will start out with a black screen and not show anything for a minute, then flash the screen a bit and should end in a solid green picture.)

viarurla, 2023-02-17:
Hi! Apologies for the delay but here is the resulting output:
```log

===log opened===
flash check:   1d62b458-b82d6850-c2-11-0  
SD test... please insert card
=============================
crc1 = 8a0b9767 crc2 = 8a0b9767   PASSED

RTC Test
========
setting clock to 2019-02-23 00:00:00
2019-02-23 00:00:00
waiting 2 seconds
new time: 2019-02-23 00:00:02
PASSED

FPGA test
=========
configuring fpga...
basic communication test...PASSED

RAM connection test
===================

Testing RAM0.1 (U501)...

Testing RAM0.2 (U502)...

Testing RAM1 (U511)...
PASSED

SD DMA test
===========
clearing RAM - loading test file - verifying -PASSED

Loading SNES test ROM
=====================

sysclk test
===========
measuring SNES clock...
21442821 Hz 
21442944 Hz diff = 123  max = 123
21441979 Hz diff = 965  max = 965
21451058 Hz diff = 9079  max = 9079
   CPUCLK: 3357861
  READCLK: 3038064
 WRITECLK: 159898
  PARDCLK: 159899
  PAWRCLK: 159898
  REFRCLK: 78500
ROMSELCLK: 479695
   CICCLK: 3579593
PASSED

SNES test status: IRQ: 00   PA: 00   MEM: 00/00
SNES test status: IRQ: 5a   PA: 5a   MEM: 5a/00

TEST SUMMARY
============

flash check:   1d62b458-b82d6850-c2-11-0  
Test      Result
----------------
SD        Passed
USB       not implemented
RTC       Passed
CIC       no run
FPGA      Passed
RAM       Passed
SD DMA    Passed
CLK       Passed
DAC       not implemented
SNES IRQ  Passed
SNES RAM  Passed
SNES PA   Passed
```

I had previously been using a custom menu purely because I only just found out they exist, but the same behaviour is exhibited regardless of that it seems. Interestingly Super Mario World exhibits a slightly different result when attempting to create a save, where there are some green bars along the top third of the screen and the music seems to simply stop altogether.

mrehkopf, 2023-02-17:
Thanks! Test seems to check out; I was going to ask if you are using a Super Nt because the 21MHz clock is moving around a lot, typical for the Zero Delay mode, then I saw your edit :) It's not an issue though.

Can you try setting your Super Nt to boot directly to cartridge and skip the startup animation? There's a slight chance of data corruption when the Super Nt holds the FXPAK in reset too long on bootup, which could also affect the code of the savestate handler.

viarurla, 2023-02-18:
Sadly no luck, really not sure why but I can live without the functionality given that everything else works fine. That said I'll still try a few things on my end but unfortunately it's a lot lower level stuff than I'm used to.

mrehkopf, 2023-02-18:
Sorry to hear that :( I have two Super Nts, maybe I can get the problem to show on either, so far I have only been using one of them (at Firmware 5.0, no JB)

redacted173, 2023-02-20:
Some other things you can try:
1) Furious' savestate firmware based on 1.10.3.  Install to a fresh SD card and run SMW (X+R to save, X+L to load).
https://github.com/furious/sd2snes/releases/download/savestate_v12/sd2snes_v1.10.3-frs-v12.zip
2) Reset SuperNT to default settings.
3) With 1.11.0 start up the first level (to the left) in SMW and immediately try to save (X+R).  Wait a few seconds and then reset to the SD2SNES menu by holding reset for several seconds.  Attach the SMW state from sd2snes/states/<ROMName>.state here.  Save executing successfully shouldn't be dependent on the majority of state, but the contents of the file may suggest how it's failing.

mrehkopf, 2023-02-20:
Now that you mention it I had a problem reported once where savestates were crashing for no apparent reason. Turned out there was a connection problem between the FPGA and PSRAM so it could only do 16-bit writes. Diagnostics didn't catch this error. Cartridge worked surprisingly well otherwise.
I added a corresponding test to the diagnostics for the new Rev.D but still need to backport it and finally roll a unified source for all revisions...

viarurla, 2023-02-20:
Gave the above a try but sadly no states ever seem to get written to the SD card at all, always comes up empty. Knowing my luck it is probably as @mrehkopf has theorised but thank you for the suggestions @redacted173 . I just hope at worst this just means the save states might be off limits but everything else will operate normally. I'll be curious about that new test when available though!

viarurla, 2023-02-20:
Hold the phone... I'm playing ALTTP and already did a normal save, and was like "might as well do a save state even though it'll crash, but who knows" and it inexplicably works... I think I might need to test more games but that seem odd to me

kencald, 2024-03-09:
> Hold the phone... I'm playing ALTTP and already did a normal save, and was like "might as well do a save state even though it'll crash, but who knows" and it inexplicably works... I think I might need to test more games but that seem odd to me

@viarurla  did you ever find a resolution? I seem to be experiencing the same issue using Super NT and SD2SNES 1.11.0 (non-pro). Attempting to create a save state causes a black screen. All other functions with my setup work fine.