mrehkopf/sd2snes #195: RevF advice

jamesgawne, 2023-02-17:
Hi I have a inherited revF board, two pads have been lifted from the SD slot, but looking at the diagram the pin outs dont match to an SD card pin out, so i am a little confussed. can I have some advice on how I can mod the pins as I have tracked them from sd_dat3 to pin55 LPC1754 and sd_cmd to pin61 LPC1754, My issue is when checking the sd card pin outs nothing matches these pin outs for the sd specs. so either im reading it wrong or the schematic is marked up wrong. 

I could just make new pads in the locations on the sd card holder but wanted to make absolutely sure its worth my time 
by putting a couple of mod wires in to see if it even works.

thx in advance

J
[image]

mrehkopf, 2023-02-17:
Hi, the SD card's pins can assume different functions based on it's operating mode (SDIO vs. SPI). The wiring you tracked is correct!

| Pin Number | SD Mode | SPI Mode |
| - | - | - |
| 9 | DAT2 | - |
| 1 | DAT3 | - |
| 2 | CMD | MOSI |
| 3 | GND | GND |
| 4 | VDD | VDD |
| 5 | CLK | SCLK |
| 6 | GND | GND |
| 7 | DAT0 | MISO |
| 8 | DAT1 | - |

jamesgawne, 2023-02-17:
Hi sorry been at work all day, thank you for the quick reply, CMD I am thinking of making a new pad from the via right next to it and will try the same from DAT3 there is a via on the way to pin 59 of the LPC infact there are 2, just my soldering skills aren't as they were 20 years ago :), But thank you for confirming the pinout for me.

regards 

J

closed 2023-06-21 by mrehkopf