mrehkopf/sd2snes #201: Early error message display unreliable

mrehkopf, 2023-02-21:
Early error messages, like "SD Card not found!", do now show reliably on Super Nt and Ultra16 consoles due to reset timing. Also, resetting the system will cause the message to not show again.

mrehkopf, 2023-02-22:
@Ramsis-SNES feel free to test ;) [issue201.zip](https://github.com/mrehkopf/sd2snes/files/10799011/issue201.zip)

Ramsis-SNES, 2023-02-22:
Good job, ikari! I tried my best but couldn't get the error message to _not_ show up with this firmware. I also like the new, more verbose error messages presented. :+1: 

Under very specific circumstances, one issue popped up though: If the sd2snes directory is missing on the SD card **and** the card is write-protected, the respective error message flickers heavily. On one occasion, parts of the system info message (with "SGB ... ok" even) appeared in the lower right corner of the screen after a few seconds. This however only happened one single time (with the flickering message, as described) and I couldn't reproduce it after 10 or so more tries.

I used multiple different SNES consoles for testing this time (Super Nt, 1CHIP Ultra16, stock US 1CHIP, US 1/1/1, US 2/1/1 and even US 2/1/2). Results were consistent across all of them. :-)

mrehkopf, 2023-02-22:
Yay, thanks for testing :) Strange issue though, could it have been a hardware glitch? (I think it's likely)
Do you remember which console it was on?

mrehkopf, 2023-02-22:
Never mind, getting the same. 😂 

https://user-images.githubusercontent.com/238664/220615811-1be4ee0a-c71f-4fd6-9b83-30daeebc73a5.mp4

Ramsis-SNES, 2023-02-22:
Right, the flickering was always reproducible, while parts of the system info message appearing was not (should have worded that part more clearly). ;-) I guess it's one and the same issue anyway, no?

mrehkopf, 2023-02-23:
Yeah, the garbage characters are artifacts of the rapid refreshing. The flickering occurred because after displaying the message, the firmware waits for the card to be removed before doing another status refresh; it does so by waiting for a status byte to become non-zero. Not taking into account that the "write protect" flag also makes it non-zero =)
Should be fixed now: [issue201-2.zip](https://github.com/mrehkopf/sd2snes/files/10818675/issue201-2.zip)
I also added printing of the hardware model and revision since that's very useful for support.

Ramsis-SNES, 2023-02-25:
Very nice, thanks! :D The flickering is gone, and I didn't encounter any other problems (tested on 1CHIP Ultra16 and Super Nt for now).

closed 2023-02-25 by mrehkopf

Absoluc4s, 2023-05-11:
This fixed my LED issue with my SD2SNES MKII Rev. F but now whenever I toggle in-game hooks, the screen goes blank for me and I have to power cycle twice for the menu to show again when it happens.

mrehkopf, 2023-05-11:
Can you confirm that this issue goes away when you revert to the previous 1.11.0 release and earlier firmwares?
I've experienced it on an old sd2snes and it was basically caused by a peak in power consumption caused by the SD Card writing throwing off CIC communications; it went away when I swapped cards.

Absoluc4s, 2023-05-11:
It doesn't happen in 1.11.0 or the earlier firmwares.

PR #202 (2023-02-25): SNES boot ROM (early error message display) fixes
fixes #201
- SNES boot ROM (early error message display) fixes
- Unify some logging (file system errors)