mrehkopf/sd2snes #204: Super Metroid hangs on 1.11.0.

sivart0, 2023-02-27:
Shortly after booting the game, it will hang.  This doesn't seem to occur with other games.
I've tested firmware 1.10.3 and an actual SM cart and the game works fine. 
I'm using an SD2SNES on an unmodded SNES.

mrehkopf, 2023-02-27:
c&p from Discord:

Do you have In-game hook and Initial holdoff activated in the in-game settings? Are you using a custom theme?

(theory: A custom theme from v1.10.3 is used with v1.11.0 so the contained hook code is incompatible, and it crashes as soon as the in-game hook kicks in after the initial holdoff.
In that case the game should not even boot if initial holdoff is disabled because the in-game hook will be active from the beginning)

mrehkopf, 2023-03-01:
Can you try the FURiOUS Savestate firmware and see if it works?
https://github.com/furious/sd2snes/releases/download/savestate_v12/sd2snes_v1.10.3-frs-v12.zip

Schmo, 2023-04-07:
For what it's worth, I have the latest firmware, my own custom themes, and I can run vanilla SM just fine on my FXPak Pro.

danbgreek80, 2023-11-19:
Recently, I played SM from beginning to end over several sessions. It froze once in the last area, just before reaching Mother Brain. I'm using a fxpak pro rev.b with firmware 1.11.0 and a brazilian snes snsm-1chip-02.