mrehkopf/sd2snes #205: Super Mario RPG Chinese translate on SD2SNES has some problem

hlkencn, 2023-03-01:
Title: Hello, I'm HLKEN, a member of SNES games Chinese Translation Team. Feedback a SA-1 chip control problem encountered in the localization process of a Super Mario RPG game.

Asked the SD2SNES firmware development team for help, our Chinese version ran into problems running on SD2SNES. During the problem location process, we suspected that the firmware of sd2snes simulated the SA-1 chip, and several interface commands may be missing. 
[use SA1 wait , like 
SA1 WAIT
LDA #$40
STA $002200

s-cpu work

SA1 START
LDA #$00
STA $002200 ]

We did some tests, and passed on multiple simulators in the Chinese version, but failed on sd2snes. [please attach a list of simulators that passed the test].

There are lots of retro game enthusiasts in China who love SNES, many of them are SD2SNES fans, and we hope to solve this problem for them. Looking forward to a response, Best Wishes.

mrehkopf, 2023-03-01:
Hi! I think if such basic things would not work then no commercial games would work at all... could you provide a small test ROM that shows the problem?
One thing to note is that the SA-1 core on the sd2snes/FXPAK Pro currently does not run at original speed but a bit slower. Could that be a cause of the problem?

redacted173, 2023-03-01:
Based on the bit in the description above for $2200 it looks like the SW emulators OR $40 (RDYB) and $20 (RESB) to keep the SA-1 in reset.  But the SD2SNES implementation only looks at $20.  It would be simple enough to OR in $40 in the two places, but is the function identical like the SW emulators assume?  Not clear.

My guess is all commercial games use $20 to get the SA-1 in and out of reset.  SMRPG does.  Is there a reason the translation can't use $20?

mrehkopf, 2023-03-01:
Ah, good find; strange implementation by the emulators. I'd like to run some test code on a real SA-1 to observe its behaviour but the official docs also state only #$20 (RESB) to setup the SA-1 for execution and run it subsequently, and to stop it again. However I guess it should be possible to pause and resume SA-1 execution (in case it's running a poll loop or similar) and resume it using #$40 (RDYB)? Though I don't see why the SA-1 would need to be paused since concurrent S-CPU access is possible at any time anyway.
Also, maybe in case of RDYB it does not heed the contents of the RV and just keeps running from where it was?

redacted173, 2023-03-01:
Yes, quite possible it's a halt as opposed to a reset.  Like you said, synchronization is best done on the fly through variables in I-RAM or BW-RAM.  No reason to halt or reset the SA-1 to accomplish this.  Both the SA-1 and S-CPU often are frequently found in spin loops on variables in those places.

As an aside, one of the hacks (Armageddon?, I think) had a bad pointer to some text which would cause the SA-1 to run through most of its address space.  For some reason S9X was able to recover, but others, including the SD2SNES, would hang.  Not sure if this was ever fixed in the hack.  That's probably not the problem encountered here.

redacted173, 2023-03-01:
Correction: bsnes treats RDYB like a halt as you suggested.  Either bit puts the SA-1 to sleep.

mrehkopf, 2023-03-01:
In a way that it starts from the reset vector again after the bit is cleared?

redacted173, 2023-03-01:
Only RESB starts from the reset vector: https://github.com/devinacker/bsnes-plus/blob/17818367f6832269006989de600e64cda0a16bf4/bsnes/snes/chip/sa1/mmio/mmio.cpp#L13

Both bits cause the SA-1 to go idle: https://github.com/devinacker/bsnes-plus/blob/17818367f6832269006989de600e64cda0a16bf4/bsnes/snes/chip/sa1/sa1.cpp#L27

That's one emulator's definition of it.  I don't think there is an official document describing it.

I'm still confused what the problem is with the translation.  I don't understand how these two major control bits are related to it.  An IPS based on some source ROM, save game srm, and explanation of expected behavior would help.

hlkencn, 2023-03-02:
In my Chinese version of Super Mario RPG game, due to the expansion of ROM, I need to pause SA1 and then switch to BANK when reading the expansion of ROM space. If SA1 reads the contents of the BANK during switching, an error will occur. This is why I suspended SA1.

hlkencn, 2023-03-02:
Because the laws of different countries are different, I don't know whether it is appropriate to send ROM links.
Link: https://pan.baidu.com/s/1z-lyp3XjaASV8Chln2UlOw?pwd=smrc
If there is a problem, please let us know.

hlkencn, 2023-03-02:
This Chinese-version game, in China, has been modified by a friend using the original SA1 game cartoon and can run the Chinese-version game normally, so I guess this operation is feasible.

hlkencn, 2023-03-02:
Please forgive me for using automatic translation tools

mrehkopf, 2023-03-02:
Thank you! Unfortunately I cannot create a Baidu account because they do not accept German phone numbers (
+49). But I'll try to find a way.

hlkencn, 2023-03-02:
> Thank you! Unfortunately I cannot create a Baidu account because they do not accept German phone numbers ( +49). But I'll try to find a way.

I can email you ROM. Please give me the EMAIL address

redacted173, 2023-03-02:
I also don't have access.  You can try the attached fpga_sa1.* bit files copied over 1.11.0.  The changes barely fit on SD2SNES mk2.  It only halts execute so you'll have to make sure all the other state machines are idle.

I'm not sure if this would actually work on a real SA-1.  Seems a spin-loop in I-RAM would be a safer solution.

[sd2snes_sa1_waitbit.zip](https://github.com/mrehkopf/sd2snes/files/10871879/sd2snes_sa1_waitbit.zip)

hlkencn, 2023-03-02:
Thank you. I will try.

hlkencn, 2023-03-10:
After testing by friends with sd2snes, this patch does not seem to solve the problem. I tried to modify the NMI of SA1 and use NMI to replace the waiting of SA1. After testing, it is normal. Thank you very much for your suggestion.

hlkencn, 2023-03-11:
After testing by friends with sd2snes, this patch  is effective. 
On the other hand I tried to modify the NMI of SA1 and use NMI to replace the waiting of SA1. After testing, it is normal. Thank you very much for your suggestion.

hlkencn, 2023-03-11:
After testing by friends with sd2snes, this patch is effective.
On the other hand I tried to modify the NMI of SA1 and use NMI to replace the waiting of SA1. After testing, it is normal. 
Thank you very much for your suggestion and help.

redacted173, 2023-03-12:
Thanks for trying it and I'm happy you were able to get the translation working on sd2snes.

closed 2023-03-13 by mrehkopf

r0g1nh0, 2024-09-09:
@hlkencn Could you send me the files with the modification? I have problems running super mario rpg on my sd2snes rev x2. The game freezes quickly after starting.