mrehkopf/sd2snes #212: Save states for Super Metroid (Japan, USA) (En,Ja) not working on SD2SNES 1.11.0

enomam, 2023-03-24:
Hey all,

I was trying to save progress on Super Metroid (Japan, USA) (En,Ja) (MD5: 21F3E98DF4780EE1C667B84E57D88675), and all the game did was transition to the pause state. I double checked a few other games (SMW, and ALTTP), and save states worked fine.

What was strange, is that the MSU-1 patched version of Super Metroid (MD5: 79A6560441CB7045E514EF71875B1FC7) did have working save states - so not sure what's going on there.

Let me know if there's any additional testing/debugging I should do on this one.

mrehkopf, 2023-03-24:
Hi, check the savestate_inputs.yml with a text editor - I think the buttons for Super Metroid are set to Select+Y+R/L.
I realize it's not a great user experience to have options hidden like this, sorry about that. I'm trying to come up with a way to make all game specific settings more obvious.

mrehkopf, 2023-03-24:
I think it might be common to pause the game while holding the R or L buttons for aiming so the combination was changed to prevent accidental saving or loading of a save state.

closed 2023-03-24 by mrehkopf