mrehkopf/sd2snes #214: Bad MSU-1 Audio with FXPak Pro (December 2022 revision)

speedyrules, 2023-03-24:
Hi
I am having issues with SNES MSU-1 Games Audio. Anyone else?
One example is Battletoads in Battlemaniacs, where the sound is all garbled.

I did try a bit around with different setups. i used:
- 1 FXPak Pro (latest revision, ordered it in december 2022 at krikkz.com), running Firmware v1.11.0
- 2 different PAL Super Nes
- 3 different video cables: multiout-composite, multiout-rgb and multiout/rad2x
- 2 tvs (LG Oled C1, hdmi / Sony Bravia Lcd, hdmi, rgb and composite)
- 3 different sd-cards: 1gb, 64gb and 128gb
- 2 msu enhanced games 
-- Battletoads in Battlemaniacs (CRC 617ac925), patched and extended (Zeldix as source)
-- Battletoads in Battlemaniacs (USA) (MSU1) [Hack by Conn & Kurrono v3].zip 

https://user-images.githubusercontent.com/25831136/227468916-5bbea036-ccb5-41b0-b2c9-80c606505b27.mp4

https://user-images.githubusercontent.com/25831136/227468997-a91386d6-3c8a-4d60-a67c-4c6c378cf373.mp4

speedyrules, 2023-03-25:
i will check 👍 
just give me some days, because i won't have access to my stuff the next few days.
but i will do as soon as i can!

speedyrules, 2023-03-26:
@mrehkopf , thank you for your patience :)

So, volume boost is set to "off".
i turned it on to try (+3.5db) and it did - as expected - not get better.
i also turned it off again and tried again, with no better results (i was hoping maybe my turning it on an off to trigger something.. but it didn't)

[image]

[image]

mrehkopf, 2023-03-30:
Weird. I guess you aren't using any pre-boosted PCM packs? I'll check the DAC timing for any quirks... if the samples are shifted by one bit they might clip in a similar fashion.

mrehkopf, 2023-03-30:
Oh, right - can you check if Super Gameboy audio is distorted in the same way?

speedyrules, 2023-04-20:
i did some more testing
- lots of msu-1 sets sound ok
- some don't: another example Top Gear (Intro Sound)

It might be that the PCMs are pre-boosted as you said.
is there any easy-to-use pcm player i can test them myself?

and i was not able to get any super gameboy msu-1 games

it's annyoing to be a noob and not being able to do more technical research, sorry :(

mrehkopf, 2023-04-24:
Hm, I did check the packs you linked and found nothing wrong with them, not pre-boosted either; certainly high amplitude but no clipping. Generally it seems like your audio signal is a bit hot 🤔 I got another bunch of PAL consoles to check if something might be happening in the audio path (e.g. a failing capacitor might happen on two consoles independently, albeit unlikely...)

mrehkopf, 2023-04-24:
It might also be worth checking this out: https://www.zeldix.net/t2068-msu-1-re-normalization-lower-pcm-volume
However even with the PCM files at full blast (but not clipping) the audio should not distort, just be very loud... 🤔

qwertymodo, 2023-04-24:
That Battletoads pack definitely looks like a boosted one.  It's not clipping, but the title track is normalized to 0dB peak, or around -9dBFS RMS, which is almost certainly way too loud.  If that pack is linked somewhere like zeldix, please comment to tell the author they need to properly normalize their tracks.  Since you said it's fine on other packs, this definitely looks like an issue of a bad pack, and not a problem with the FXPAK.

mrehkopf, 2023-04-24:
Still I tried the same pack and don't get any distortion. Maybe my audio path is more forgiving (like I'm using a mixer with 10dB of headroom I think...)

qwertymodo, 2023-04-24:
Yeah, the fact that there's no clipping in the actual files means it's probably possible to tame it down with attenuation in your audio chain, but they definitely don't look correct to me.

speedyrules, 2023-04-25:
thanks for all the input!
so, i will try:
- another snes (if i can find one.. even as we said it would be a big coincidence having the same broken capacitors on 2 sness)
- audacity import and playback
- normalize with the exe file from zeldix (link above)
- contact the zeldix thread directly
i'll be back :)

speedyrules, 2023-04-25:
furhter reply: 
Relikk
[Re: Battletoads in Battlemaniacs](https://www.zeldix.net/t1693p75-battletoads-in-battlemaniacs#43288)

    Reply with quote

Post Today at 18:06 by [Relikk](https://www.zeldix.net/u917) (Online)
The amp is purely speculation, as I have no technical knowledge of the workings of it where an SD2SNES is concerned, but it seems like it's sensitive to higher amplitudes.

I have an SD2SNES version of the Pro. It'd be one of the first run Pro's that wouldn't fit into a US cartridge shell, as I got it when it was originally announced.

mrehkopf, 2023-04-25:
Hmm, I guess to rule out or confirm a bad DAC or amp on the FXPak some measurements would need to be done...
I think I can rule out a DAC timing bug; I just shifted all samples left by one bit and the distortion is much much worse even than your initial recordings.

speedyrules, 2023-04-25:
how would i make this measurements on the DAC or amp?

mrehkopf, 2023-04-26:
It would involve taking a multimeter or oscilloscope to some points on the FXPak PCB while it's running - the top shell of the SNES also needs to be removed to be able to reach the points (unless you have some sort of plug-in adapter for playing imports).

speedyrules, 2023-04-26:
ufff.. i am afraid this is out of my reach of abilities :(

i will try the different power bricks though ..and hope there will be a difference.. 

but else if that does not help, what is your suggestion? 
contact krikkz and and for repair?

mrehkopf, 2023-04-26:
If you live in the EU I could take a look at your FXPak personally, probably easier and faster than shipping to and from krikzz.
However we should explore the other possibilities first so please try your other power supplies and I'm going to rig up a SNES to simulate a weak 9V DC supply / failed capacitor.

speedyrules, 2023-04-26:
+1
i live in switzerland :)

- i will check different original power bricks
- and just found another pal snes, with cables and power brick. so i will try that one soonish too

will report after i got that additional snes

mrehkopf, 2023-04-27:
Hmm, Switzerland is geographically convenient but I'd still need to pay import tax when you send it to Germany 😅
I tinkered around a bit yesterday and could provoke the same distortion when running the audio amp rail (which is supposed to be 12V, not 9V, sorry) at around 3.5V. I haven't been able to reach that low a voltage even by removing a bunch of capacitors entirely so I'm wondering if it's even possible to reach that voltage by "natural" causes like aging.
Some other failure modes are possible, like a specific transistor or one of 2-3 resistors failing but how likely is that to happen on two independent consoles?
So my current bet is on the power supply 🤞 Looking forward to your results :)

speedyrules, 2023-05-01:
back amd reporting :)

i did a few things:
- i managed to get an extra PAL snes
- i got some power bricks and measured their output
- created a small SD card just to check the issue i am having
- checked all the power bricks with all the snes
- i checked the pcm files also now on foobar with pcm plugin
- checked import tax issues

let me begin:

now i have 3 PAL snes to test:
A. my yellow one: SNSP-001A (FRG), input AC9V 1.3A, power brick to use NES-002E
B. my mint one: SNSP-001A (FRG), input 9V~ 1.3A, power brick to useNES-002E
C. my buddy's one: SNSP-001A (UKV), input AC9V 1.3 A, power brick to useNES-002-GBR

[image]

the power bricks:
1. SNES power brick, NES-002EU, output 9V~1.3 A, measured 10.8V
2. SNES power brick, NES-002E, output AC9/9.4V 1.3A, measured 10.7V
3. NES power brick, NES-002ED, output AC 9V 1.3A, measured 11.3V
4. NES power brick, NES-002ED, output AC 9V 1.3A, measured 11.3V
5. NES power brick, NES-002ED, output AC 9V 1.3A, measured 11.3V
6. SNES power brick, NES-002E, output AC9/9.4V 1.3A, measured 10.7V

[image]
[image]
[image]

I copied Battledoads and the firmware on a new 1GB SD-Card.
I also found another game with issues: Killer Instinct (Arcade Version): This one has video issues on the left of the screen.
Also copied on that SD Card to test.

Results:
- Console A with all the 6 power bricks: NOK
- Console B with all the 6 power bricks: NOK
- Console C with all the 6 power bricks: NOK

Then i installed foobar2000 (32-bit) with the pcm plugin to check the pcm:
same result as with audacity: the pcm plays well.

last but not least, i did check about import taxes: there are none to pay if it is declared properly as "repair" or "return":

https://www.bazg.admin.ch/bazg/de/home/information-private/interneteinkauf--post--und-kuriersendungen/einfuhr-in-die-schweiz/ruecksendung--reparatur--veredelung.html

oh.. and i also consulted chatGPT: here is its anwer: :)
Hello! I'd be happy to help you with your question.

Super Nintendo consoles can experience a variety of failure scenarios, including sound issues. The likelihood of specific components failing on two independent consoles can vary based on a number of factors, such as age, usage, and environmental conditions.

However, it is not uncommon for electronic components to fail over time due to various factors such as heat, voltage surges, and general wear and tear. Some components, such as transistors and resistors, are more prone to failure than others. In addition, manufacturing defects can also contribute to failures.

That being said, it's important to note that the likelihood of two independent consoles experiencing the exact same failure scenario is relatively low, especially if the consoles were manufactured at different times or in different locations.

If you are experiencing sound issues on your Super Nintendo console, it's best to troubleshoot the issue by checking the console's connections, trying different cables or TV inputs, and testing the console with different games. If the issue persists, it may be necessary to have the console professionally repaired or replaced.

speedyrules, 2023-05-01:
looking forward to your results and opinion :)

mrehkopf, 2023-05-10:
👀 very thorough testing, really doesn't leave anything to be desired. Thank you 🙏
As expected after seeing your results, Variac test did not lead to any new conclusions. 😅 I was able to lower the line voltage to about 160V (~7V secondary) at which point the console started crashing, without any changes in audio.
It must be the FXPak.
Soo hit me up on Discord or Twitter I guess and we can discuss the further details :)

speedyrules, 2023-05-13:
Contacted you on discord 👍

mrehkopf, 2023-06-16:
Turned out to be a broken op-amp IC. Works great after replacing a 20ct part. ;)

speedyrules, 2023-08-10:
And we might have found an issues with the "last run super nt"
Thanks @mrehkopf for the awesome help!

closed 2023-08-10 by speedyrules