mrehkopf/sd2snes #216: Save states don't work well in Chrono Trigger

FoxLisk, 2023-04-03:
Does anyone know why this is? As far as I can tell, it doesn't use an expansion chip. Loading a save state often immediately messes up graphics, even just saving a state and then loading it without doing anything in between. I'm also getting stuck on just like a full orange screen sometimes.

redacted173, 2023-04-11:
Try the attached fpga_base.bi* files by copying them into your SD card's sd2snes/ directory over the 1.11.0 files of the same name.

Chrono Trigger uses the PPU $2134 register as a source to zero out WRAM using a DMA.  The CTX state machine captures the WRAM write then immediately overwrites the shadow address with the PPU register.  This means we lose the zeroing of WRAM in the save state.  These locations for WRAM are used for indirect HDMA source data to PPU registers which causes graphics to mess up when the save state is loaded.  I'm sure this isn't the only game which does this kind of DMA so hopefully others are working now, too.  The fix blocks the PARD snoop when there is a write.

[sd2snes_ctx_blockpardonwrite.zip](https://github.com/mrehkopf/sd2snes/files/11195516/sd2snes_ctx_blockpardonwrite.zip)

FoxLisk, 2023-04-11:
@redacted173 heyyy it seems to work! I don't have time to test it super thoroughly now, but the very simple stuff that was crashing before no longer does. Thanks so much!

Will this make it into the official sd2snes release at some point?

closed 2023-04-11 by FoxLisk