mrehkopf/sd2snes #217: Miscolored SGB boarder after soft (in game hook) reset

CrispyBuns, 2023-04-05:
After resetting Pokemon silver (MSU1) using the in-game hooks the SGB Border gets miscolored in the prosess

redacted173, 2023-04-29:
Are you using the latest version of the patch?  r2021.04.09?

Does it happen for the unmodified (non-MSU) ROM?

Are you using an unmodified sgb_snes.bin files?

Does it always happen when you reset (L+select+start+R) or just in certain cases?

I tried resetting with the in-game buttons from the title screen and from inside the game in the both the unmodified and MSU ROM and the border colors always look fine.  The only thing I noticed is a few second delay for reset, maybe because of the save RAM?, on the MSU version.

CrispyBuns, 2023-06-14:
Sorry for the late replay Yes I am using an unmodified sgb2_snes.bin. and No I haven't tested with Non MSU-1 games. and yes it only happens when I to a in-game hook reset