mrehkopf/sd2snes #218: SuperCIC not switching 50/60hz

msimplay, 2023-04-09:
I upgraded to FX PAK from an older SD2SNES cart and almost everything is working.

However I did notice that SuperCIC is not working the way it did on the old cart it's actually doing nothing no matter what the setting is in the options menu.

I can imagine most people don't care about 50hz but it's kinds of nostalgic for me so it would be nice if the SuperCIC option in the cart works as intended

Romain80200, 2023-04-10:
Hello, please try the file in this thread:
https://github.com/mrehkopf/sd2snes/issues/186

It solves the problem with cic and the missing leds.

msimplay, 2023-04-10:
Yes that totally fixed the issue thanks very much

mrehkopf, 2023-04-10:
Thanks. Still busy with family obligations due to Easter break.

closed 2023-04-10 by mrehkopf

Romain80200, 2023-04-10:
Family first.

jamieshaw, 2024-05-01:
Apologies to dig up an old thread, but I've got an SD2SNES that exhibits the same issue with 1.11.0, and having tried the "issue182-try2" firmware, can confirm it does bring back the cart LEDs, but only _sometimes_ properly sets the video mode on the SNES.

There doesn't appear to be any specific pattern that allows me to get into pair mode, just sporadically when turning on the SNES, the light will appear green for PAL (SNSP-CPU-02 console), and it'll work as expected. Maybe 1 in every 20 power-ups.

Most of the time, powering on results in an NTSC signal and appears to not react to pair mode.

jamieshaw, 2024-05-02:
Just to debug further, have tried FW 1.10.3, as well as both variants of "issue186" FW to no avail — the problem is still intermittent.

It seems that for the most part, if it does boot into PAL 50Hz mode, the menu is garbled until reset.  Wondering if it's a timing issue or dodgy SD2SNES cart, it _isn't_ a FXPAK Pro (but now potentially thinking about investing…) 🤔

msimplay, 2024-05-02:
Hmm it depends on your installation of the SuperCIC.

I've had a newer version of SuperCIC snes work fine with new FXPAK pro.

However my old Snes with Super CIC.
Only works properly with Sd2snes.

Also with FXPAK pro I've had some issues with compatibility of some games not working.
I don't know for what reason but i went back to using Sd2snes because I had 2 Snes's with Super CIC and I only needed one.

The main difference is that on fxpak you can use msu1 on SA1 chip games.

But the code base on the old sd2snes is very mature its very stable and works with every game I have thrown at it.

Fxpak is relatively new.

This my personal experience but I thought to just let you know to not sell your sd2snes straight away if you want to buy th Fxpak

jamieshaw, 2024-05-02:
> Hmm it depends on your installation of the SuperCIC.

It's the [SuperCIC AIO board](https://www.retroupgrades.co.uk/product/snes-multi-aio/).

And the SD2SNES I have is an AliExpress clone, which probably doesn't help — but then being open source and all, I thought I'd take the plunge.

Ideally, I'd love to be able to narrow down if it's a software bug, or a hardware issue, as it's easy enough for me to return the SD2SNES and look at alternatives. But I'd begrudge spending twice the amount for a FXPAK Pro, and hit the same problem, or as you say, _other_ issues.

msimplay, 2024-05-02:
There's numerous revisions of the original Sd2snes or I should say fixes.

Now that its a clone I wonder which version of the sd2snes it cloned.

I have an early revision that has numerous issues but issues were worked around in firmware a lot of the time.