mrehkopf/sd2snes #222: Jurassic Park (US) First Person View is Extremely Slow

majorpbx, 2023-05-18:
I'm not sure what else to include in this submission but here is the basics. Playing Jurassic Park (US) when you go into a building the rendering becomes extremely slow. I don't believe this game has a special chip and I don't have any odd settings. I am running the latest public release of the OS. If there are any other details I can provide please let me know.

redacted173, 2023-05-20:
Try disabling savestates in the SD2SNES menu:
X->Configuration->Savestates Settings->In-game savestates->Off

If that doesn't help then try disabling in-game hooks:
X->Configuration->In-game Settings->In-game hook->Off

I think Ikari has plans to rewrite parts of the savestate code which should address this problem.

majorpbx, 2023-05-20:
Ok, I think save states is the only thing enabled but I'll check both settings. I'll let you know what i find and ix it helps.

Thank You

majorpbx, 2023-05-21:
Well In-game hook was off but save states was on. Turned save states off and it worked like a charm. Thank You