mrehkopf/sd2snes #226: Building using MK3 config is currently broken

codemann8, 2023-06-21:
It seems `CONFIG_MK3_STM32 = n` was added to config-mk3 but uart.h is sending it down the wrong path with this line: `#if defined CONFIG_MK3_STM32`

However, even after fixing that line, there are places later on that also seem to fail.

mrehkopf, 2023-06-21:
Weird, I'm compiling with CONFIG=config-mk3 on a regular basis. conf2h.awk should not output any `#define`s for properties set to `n`. What does your obj-mk3/autoconf.h look like? There should be something like this near the top:
```
// CONFIG_MK2 = n
#define CONFIG_MK3
// CONFIG_MK3_STM32 = n
```

codemann8, 2023-06-21:
I don't seem to have that file anywhere

codemann8, 2023-06-21:
Oh...this: https://github.com/mrehkopf/sd2snes/blob/f5c28d83a884f317c32a7c8205f716e91dceceef/.gitignore#L19

mrehkopf, 2023-06-23:
I tried the Docker image on an x86_64 Debian bullseye and it built without problems. I used the prepared build.sh from the repository.

codemann8, 2023-06-24:
As in you used the build.sh from the repo instead of the build.sh from the docker repo?

mrehkopf, 2023-06-24:
Like this:
```
user@host:~/src$ git clone https://github.com/alttpo/sd2snes-build-docker.git
Cloning into 'sd2snes-build-docker'...
remote: Enumerating objects: 38, done.
remote: Counting objects: 100% (38/38), done.
remote: Compressing objects: 100% (29/29), done.
remote: Total 38 (delta 15), reused 28 (delta 7), pack-reused 0
Receiving objects: 100% (38/38), 4.85 MiB | 22.00 MiB/s, done.
Resolving deltas: 100% (15/15), done.
user@host:~/src$ cd sd2snes-build-docker/
user@host:~/src/sd2snes-build-docker$ sudo ./build.sh
Cloning sd2snes repository into sd2snes/
Cloning into 'sd2snes'...
remote: Enumerating objects: 12946, done.
remote: Counting objects: 100% (1772/1772), done.
remote: Compressing objects: 100% (555/555), done.
remote: Total 12946 (delta 1272), reused 1661 (delta 1211), pack-reused 11174
Receiving objects: 100% (12946/12946), 14.81 MiB | 7.40 MiB/s, done.
Resolving deltas: 100% (9294/9294), done.
Building docker image which compiles the sd2snes firmware and FPGA
[+] Building 3764.9s (23/23) FINISHED
 => [internal] load .dockerignore                                                                                                   0.5s
 => => transferring context: 2B                                                                                                     0.1s
 => [internal] load build definition from Dockerfile                                                                                0.6s
 => => transferring dockerfile: 2.50kB                                                                                              0.2s
 => [internal] load metadata for docker.io/stronglytyped/arm-none-eabi-gcc:latest                                                   1.8s
 => [internal] load build context                                                                                                   3.9s
 => => transferring context: 48.18MB                                                                                                3.2s
 => CACHED [install 1/4] FROM docker.io/stronglytyped/arm-none-eabi-gcc:latest@sha256:0a5cf4a7122848d8e1b677695f0ae16891d76a018b82  1.2s
 => CACHED [install 2/4] WORKDIR /work/quartus                                                                                      0.0s
 => [install 3/4] RUN wget https://download.altera.com/akdlm/software/acdsinst/20.1std.1/720/ib_tar/Quartus-lite-20.1.1.720-linu  338.8s
 => [install 4/4] RUN tar xf Quartus-lite-20.1.1.720-linux.tar &&  ./setup.sh --mode unattended --accept_eula 1 --installdir /op  835.0s
 => [stage-1  2/15] COPY --from=install /opt /opt                                                                                 862.4s
 => [stage-1  3/15] WORKDIR /work                                                                                                   2.1s
 => [stage-1  4/15] RUN apt update && apt install -y gawk unzip curl libglib2.0-0 libtcmalloc-minimal4                            110.5s
 => [stage-1  5/15] RUN curl https://getmic.ro | bash ; cp micro /usr/local/bin/                                                   15.3s
 => [stage-1  6/15] RUN mv /opt/intelFPGA_lite/20.1/quartus/linux64/libboost_system.so /opt/intelFPGA_lite/20.1/quartus/linux64/li  3.2s
 => [stage-1  7/15] ADD ./sd2snes /work/sd2snes                                                                                    18.9s
 => [stage-1  8/15] WORKDIR /work/sd2snes                                                                                           1.5s
 => [stage-1  9/15] ADD ./verilog-settings.mk verilog/settings.mk                                                                   1.3s
 => [stage-1 10/15] RUN cd ./utils && make                                                                                          7.2s
 => [stage-1 11/15] RUN cd ./src/utils && make                                                                                      2.7s
 => [stage-1 12/15] RUN cd ./verilog/sd2snes_base && make mk3                                                                     490.1s
 => [stage-1 13/15] RUN cd ./verilog/sd2snes_mini && make mk3                                                                      65.6s
 => [stage-1 14/15] RUN sed -i '27 i #include <ctype.h>' ./src/sgb.c                                                                3.0s
 => [stage-1 15/15] RUN cd ./src && make CONFIG=config-mk3                                                                         22.2s
 => exporting to image                                                                                                            443.8s
 => => exporting layers                                                                                                           443.0s
 => => writing image sha256:e9ebb51b6f03d843e8e5c7fd4316e7a06a58fc932e1af5622d884672be3ffb17                                        0.3s
 => => naming to docker.io/library/sd2snes-fpga                                                                                     0.3s
Creating docker container from the built image to extract im3 files
Extracting im3 files
Successfully copied 146kB to /home/user/src/sd2snes-build-docker/.
Successfully copied 169kB to /home/user/src/sd2snes-build-docker/.
Successfully copied 58.9kB to /home/user/src/sd2snes-build-docker/.
Removing temporary container
ead75e5876c0e2ad8658994a807b9f016741228682b4b99ba9ff3abf36d5af4b
Successfully built firmware and FPGA images for MK3
```

codemann8, 2023-06-25:
Alright, figured out the issue. Turns out that downloading that docker repo's source as a zip file from GitHub's site was stripping out something. Running git clone from MinGW seems to have resolved it.

closed 2023-06-25 by codemann8