mrehkopf/sd2snes #228: SPC7110 Support

mrehkopf, 2023-06-30:
Implement SPC7110 support

bitrot-alpha, 2023-10-08:
It's disappointing that the MiSTer project uses an incompatible software license (GPLv3). They have an SPC7110 core already implemented..

https://www.gnu.org/licenses/gpl-faq.html#v2v3Compatibility
https://github.com/MiSTer-devel/SNES_MiSTer/tree/master/rtl/chip/SPC7110

actraiser30, 2024-06-04:
Still waiting for far east of Eden to be playable on fx pack pro....

odiaboeeu, 2025-05-05:
Well, let's ask the code owner @srg320 if he would give permission for someone to port to sd2snes, because it would be wonderful if he did.

dderby, 2025-05-05:
> Well, let's ask the code owner [@srg320](https://github.com/srg320) if he would give permission for someone to port to sd2snes, because it would be wonderful if he did.

It's possible that even with explicit permission from the code author, this may not be enough to override the legal constraints of software licence incompatibilities.  I would not recommend this approach without seeking legal advice first.

> It's disappointing that the MiSTer project uses an incompatible software license (GPLv3). They have an SPC7110 core already implemented..

Whilst it is true that GPLv2 and GPLv3 have incompatibilities, a GPLv2 project can use the ["version 2 or later" SPDX licence identifier](https://spdx.org/licenses/GPL-2.0-or-later.html) to allow for GPLv3 code to be used within a GPLv2 project.  This is even stated in [the link you provided](https://www.gnu.org/licenses/gpl-faq.html#v2v3Compatibility), and is further evidenced in [GNU's licence compatibility matrix](https://www.gnu.org/licenses/gpl-faq.html#AllCompatibility).  RMS also [wrote up a post](https://www.gnu.org/licenses/identify-licenses-clearly.html) on using explicit SPDX identifiers to clarify the difference between `GPL-2.0-only` and `GPL-2.0-or-later`.