mrehkopf/sd2snes #230: 1.10.3 OK 1.11.0 KO  (Very similar to 184)

jjnetcn, 2023-07-15:
Me and https://github.com/mrehkopf/sd2snes/issues/184 has Almost the same problem
Our test_ Log.txt is very similar, except for the slightly lower clock
His https://pastebin.com/xY7JmDLn My https://pastebin.com/ca5Dsm99

I have tried v.10.3. frs https://github.com/furious/sd2snes/releases/tag/savestate_v12 And USB 2snes_ V11,
https://github.com/RedGuyyyy/sd2snes/releases/tag/usb2snes_v11 They can all work

The main purpose of i upgrading from 1.10.3 to 1.11.0 is to use the savestate function,
but v.10.3. frs does not work well, and there are often problems when loadstage after save,
nd it also damaged my saves xxx.SRM file.

I have tried the firmware v11.0, 1.11.0b2-RevD, v1.11.0b1 that I can find, all likes v.11.0
there are all frozen when the main menu appears,

According to the post, https://github.com/mrehkopf/sd2snes/issues/161 In the environment of v11.0,
When replacing the sampling's base. bit, the system starts with a black screen,
When replacing the sampling1 or sampling2, now can enter the main menu, 
but most games cannot start normally (some cannot start, some only have sound and no images, while v.10.3 is fine)

Here is the picture of my sd2snes device:
https://ibb.co/NpBWSC9
https://ibb.co/tDNY2rx
https://ibb.co/CBTqqQz

How should I solve it the problem? thanks advanced

mrehkopf, 2023-09-14:
Hi, sorry for the late reply. Your issue sound much the same like #184 indeed. I found that my new sampling method for stability improvement could have adverse effects on some consoles. I do not know why since I don't own such a console but I will prepare an FPGA bit file for troubleshooting shortly to see if the old method works better for you.

mrehkopf, 2023-09-15:
That said, I looked at your pictures and it seems you have a clone cartridge. So in your case it might be the cartridge.

mrehkopf, 2023-09-17:
https://sd2snes.de/files/sd2snes_v1.11.0-testsampling.zip
Can you extract this over your current v1.11.0 installation (so that fpga_base.bi3 and fpga_base.bit are overwritten) and try if it works better?

closed 2026-05-16 by mrehkopf