mrehkopf/sd2snes #232: bahamut lagoon hack translation not working

gaialily, 2023-07-20:
Hi,
A translation hack, in french, for bahamut lagoon doesn't work on my sd2snes. I have the 1.11 firmware.
This is this hack : https://www.bad-company.fr/fr-fr/traductions/super-nintendo/bahamut-lagoon/
They say in the readme that they created a special software to translate the game.
Though, another translation hack, from another group, outdated because finished only at 90% (and not update from 2001), works on my sd2snes.
Could it be the software used ? 
Could you help me ?

Thanks.

Ramsis-SNES, 2024-09-02:
Hi,

in case you haven't figured it out by yourself in the meantime, it's an issue with the translation, not the sd2snes/FXPAK. The translation overwrites the internal game title with one that's 1 byte too long, messing up the ROM makeup byte at file offset $FFD5 in the process.

You can easily fix it yourself, though. Open the patched ROM in a hex editor, navigate to $FFD5, and change the byte at that location (should be $32, which causes the game not to work) to $31. Don't forget to save the file, and copy it to your SD card.

This is what it should look like with the fix applied:

[image]

The translation should now work just fine. :-)

Please remember to close this issue once you've seen this. Thanks!

gaialily, 2024-09-03:
Wow ! Thank you very much ! It works !

closed 2024-09-03 by gaialily