mrehkopf/sd2snes #234: [Hardware] Aliexpress clone cart: RTC not ticking

bitrot-alpha, 2023-07-31:
Hi,

Sorry in advance if you don't support Chinese clones.

Board info
> FW: 1.11.0
> SD Card: No-name chinese 16GB
> Revision according to seller page: X (Whatever that means..)

I bought a clone cart from Aliexpress, and it appears to work fine except that the RTC does not tick when the cart is powered off. It seems to hold whatever time I set it to, but it resets to that exact set time when the cart is powered off or SNES reset is triggered.

The test firmware showed no problems with the cartridge except for the RTC.

I took apart the cartridge to check the battery voltage, and it is at 3.0 volts (Not a CR2032, they used the smaller CR1220 instead). I tried removing the battery and waiting 30 seconds before re-installing it. Is there some other troubleshooting steps I could do? I can send photos of the board.

I am somewhat proficient in SMD soldering. Would replacing the RTC chip do anything useful?

Test firmware log:
```
===log opened===
derp
SD test... please insert card
=============================
data=111133b04504
crc1 = d805a1bf crc2 = d805a1bf   PASSED

RTC Test
========
setting clock to 2011-01-01 00:00:00
2011-01-01 00:00:00
waiting 5 seconds
new time: 2011-01-01 00:00:00
FAILED

FPGA test
=========
configuring fpga...
fpga_pgm: 212501 bytes programmed
FPGA configured
basic communication test...PASSED

RAM test
========
Testing RAM0 (128Mbit) - clearing RAM - writing RAM - verifying RAM -RAM0 PASSED

Testing RAM1 (4Mbit) - writing RAM - verifying RAM...RAM1 PASSED

SD DMA test
===========
clearing RAM - loading test file - verifying -PASSED

Loading SNES test ROM
=====================

sysclk test
===========
measuring SNES clock...
21478008 Hz 
21478008 Hz diff = 0  max = 0
21478008 Hz diff = 0  max = 0
21478008 Hz diff = 0  max = 0
   CPUCLK: 3420396
  READCLK: 3094644
 WRITECLK: 162876
  PARDCLK: 162876
  PAWRCLK: 162876
  REFRCLK: 78735
ROMSELCLK: 488628
PASSED

[... 509 lines cut ...]
SNES test status: IRQ: 5a   PA: 5a   MEM: 01/fd
SNES test status: IRQ: 5a   PA: 5a   MEM: 01/fe
SNES test status: IRQ: 5a   PA: 5a   MEM: 01/ff
SNES test status: IRQ: 5a   PA: 5a   MEM: 01/00
SNES test status: IRQ: 5a   PA: 5a   MEM: 5a/00

TEST SUMMARY
============

flash check:   1111-4-33b04504-dfb09a7f-08-11-0  
Test      Result
----------------
SD        Passed
USB       not implemented
RTC       FAILED
CIC       no run
FPGA      Passed
RAM       Passed
SD DMA    Passed
CLK       Passed
DAC       not implemented
SNES IRQ  Passed
SNES RAM  Passed
SNES PA   Passed
```

bitrot-alpha, 2023-07-31:
Board photos

[image]
[image]
[image]

bitrot-alpha, 2023-07-31:
Aliexpress seller suggested I solder the USE_BATT jumper (JP402 on this version of the board) then set the time, and then unsolder the jumper. This didn't do anything useful.

mrehkopf, 2023-07-31:
USE_BATT has nothing to do with that, in fact it has no effect at all. Looks like the seller has no idea what they're talking about 😁 RTC failure seems commonplace for those carts, I once fixed one by replacing the load caps on the RTC crystal (C408+C409), they're supposed to be 15pF but you might want to try 12pF or 10pF as well.

But in all honesty it should be the seller's responsibility, not yours nor mine ;)

mrehkopf, 2023-07-31:
Also yikes, what's with the cracked coil on the switching regulator? 😬

bitrot-alpha, 2023-07-31:
Thanks for getting back to me! I'll try ordering some SMD caps I guess. 0608 size?

They shipped the cart in a pretty thin bubble mailer. When I opened the cart up for the first time, I wondered why a tiny little piece fell out, turns out it was that missing chunk of the coil haha.

mrehkopf, 2023-07-31:
Looks like 0603.
Oh dear. Good luck. 🥲

bitrot-alpha, 2023-08-08:
Received my replacement caps today. I tried the 10pf and that didn't fix it.

15pf did work though! I wish I had an ESR meter to check the ones I removed..

closed 2023-08-08 by bitrot-alpha