mrehkopf/sd2snes #244: SA-1 Timing Regression in 1.11.0

GooberScotsman, 2023-11-20:
I was testing out this version of the firmware and noticed some slowdown during parts of Super Mario RPG. The earliest tell was the opening when Peach is captured by Bowser. When the butterflies leave, the game slows down and desyncs with the music. Mario is suppose to jump with the music sting but now jumps after. I tested with a real Japanese cartridge to confirm my findings and there was no slowdown with the butterflies. Firmware 1.10.3 doesn't have this issue when tested on the SD2SNES, but 1.11.0 does. I even tested 1.11.0 on the FX Pak Pro and the same slowdown issue occurred. This is present when playing both Japanese and North American versions. I currently don't know what other SA-1 games/hacks may be affected at this time.

redacted173, 2023-11-30:
There was a post regarding this on a forum: https://krikzz.com/forum/index.php?topic=12501.0

mrehkopf, 2025-06-20:
This is a regression caused by introduction of the more robust bus strobe sampling (SNES_PULSE generation), leaving fewer time slots for the SA-1 apparently. A quick fix reverting the PULSE logic for the SA-1 core proved unreliable - games running in FastROM (*before* enabling the SA-1) would crash. Closer investigation is underway.