mrehkopf/sd2snes #248: Character Conversion DMA issues on SA1

blackmagemathie, 2024-03-12:
upon upload/conversion using CCDMA (type 1 (BW-RAM to I-RAM), 2bpp mode, VRAM horizontal character size set to 2^0), graphics are garbled and/or duplicated [(see forum thread for more details)](https://krikzz.com/forum/index.php?topic=12880.0).
- in-game hooks, savestates, etc. were all disabled during testing
- issue occurs on more than one flashcart/console
- issue occurs with more than one CCDMA system
- randomised writes/reads test suggests issue occurs during actual Character Conversion (or when setting it up), and Bitmap Access Area is not responsible
- further tests suggest issue may be buffer-related

mrehkopf, 2024-03-13:
Thanks for the greatly detailed report on the forum and test ROMs. Hope to be able to look into it soon :)

mrehkopf, 2026-11-13:
For single character width, the next row's address was ready too late because the next address is computed in the same clock as the current one is loaded into the source address register. In single character mode there are no steps inbetween because the next row is needed immediately, so the source address register was always reloaded with the value from the row before, resulting in a character order of 0, 1, 1, 2, 2, 3, 3, etc.

To fix this for single character mode, the address register simply isn't reloaded at all per row because the byte increment naturally puts it on the start address of the next character already.