mrehkopf/sd2snes #250: Super game boy 2 doesnt work.

zarkon0, 2024-03-24:
Hi.

In my sd2snes rev. X2 super game boy 2 doesnt work. The screen come
to pink when I try to load a gameboy game.

Super Game boy 1 works fine.

zarkon0, 2024-05-06:
Hi.

Yes, it's a PAL console with the "region free" and 60Hz mods.
I've tried all combinations (pal/50Hz, pal 60Hz, region free 50Hz and region free 60Hz) and it doesn't work.

actraiser30, 2024-06-04:
Is sd2snes rev. X2 a chinese clone??

zarkon0, 2024-06-04:
My sd2snes rev. X2 is made in China.

actraiser30, 2024-06-04:
It's a Chinese clone that's why sgb2 not working try sgb1
In the original is working fine.

jamieshaw, 2024-06-05:
My experience using SGB2 was also using a Rev. X cart and had no issues. That said, variance between Chinese "clones" is going to vary.

That said, I had other issues regarding SuperCIC pair mode and have since moved to an official FXPAK PRO. So yeah, quality amongst carts is not great.

DoctorDan1986, 2024-06-05:
Well....the harsh truth is that you get what you pay for.  Sometimes the clones are just as good, but usually, clones run the gamut from "pretty okay" to "steaming pile of moist garbage".Sent from my iPhone

actraiser30, 2024-06-05:
I wouldn't recommend the clone unless someone has less money,I would go for X6 anyway,the quality is very low you don't know what you buying by getting a clone.the open source does not mean anything since it uses the os from the authentic sd2snes.

mrehkopf, 2026-05-16:
I'll still call them clones any day, even the Mk.2 clones where the hardware design was still published.
Their makers just take and give nothing back. They roll their own designs, never communicate any changes, and I'm expected to magically support every variant they throw out there. Not possible I'm afraid :)

closed 2026-05-16 by mrehkopf