mrehkopf/sd2snes #252: Firmware 1.11.0 Jurassic Park 2 sound corrupted

kscorpionx, 2024-04-13:
After playing 30 minutes or more with the sd2snes and then playing Jurassic Park 2 The sound of the plane in the intro sounds corrupt, this does not happen if you play Jurassic Park for the first few minutes, it was tested with Snes 2chip and a Snes mini.

Ramsis-SNES, 2024-09-03:
> After playing 30 minutes or more with the sd2snes

You mean, 30+ minutes of _any game?_ Or 30+ minutes of _Jurassic Park 2_, and then resetting the game so it starts from the beginning?

What sd2snes hardware revision do you have? Do you play in 50 or 60 Hz (or PAL/NTSC)? Does the ROM region match your video mode? Also, could you make a video of the scene with the glitched sound effect for reference?

kscorpionx, 2024-09-04:
After playing 10 minutes with any game this problem occurs but I found that if you don't press start and let the intro repeat a few times this problem doesn't happen, I am playing on 60mz ntsc and yes this rom matches my region , I am using an sd2snes. made by ky technologies rev x, i know this is chinese but i want to buy a krikzz so i want to know if this doesn't happen with that one. Here is the video with the failure, the plane sounds distorted.

https://github.com/user-attachments/assets/9476b5c7-d183-4aa1-a9db-909c4aa12ec6

Ramsis-SNES, 2024-09-04:
Sorry, I can't seem to reproduce it.

Then again, I don't know what games you were playing other than the one in question, and how you even got back to the menu (soft reset? Console reset? Power off/power on?). For me, there's too much guesswork involved and too much time to be invested trying again and again, so I'm afraid I'll call it quits on this one.

To be perfectly honest with you, I don't even know I'm actually hearing what you call a "failure" in your video. The sound effect sounds a bit boomy, but I wouldn't really call it a glitch in comparison to my own tests before I would have heard what it sounds like on your setup when it's _not_ glitching out. (You might want to make another video of that but I don't really think it's worth the effort, unfortunately.)

closed 2025-02-05 by kscorpionx