mrehkopf/sd2snes #253: Multiple problems with 1.11.0 version

ZarkonPro, 2024-05-08:
Hi.

I've detect this problems:

- The green led doesn't bright. The option menu to configurate the led bright level does nothing.
- Randomly the screen turns to black. The sound continues working. I've deselected all options of in-game hooks, cheats, patch region...
- Super Game Boy 1 is very unstable. Some games doesn't work (black screen),. Others works after press the reset buttón (I've selected that reset button returns to menu, but in this cases, magically, it doesn't return to menu, it returns to Super Game Boy and then works).

actraiser30, 2024-06-05:
Hi you have an original sd2snes or fxpack or a clone?

Ramsis-SNES, 2024-08-13:
> * The green led doesn't bright. The option menu to configurate the led bright level does nothing.

cf. #186.

Feel free to try [the ZIP file ikari provided](https://github.com/mrehkopf/sd2snes/issues/186#issuecomment-1434480057).