mrehkopf/sd2snes #255: Firmware 1.11.0 Stunt Racer FX

kscorpionx, 2024-06-13:
The red LED is active from the start to the end of a race, this does not happen in the menus.

Ramsis-SNES, 2024-08-13:
> There are occasions where **games use the battery SRAM area as work RAM.** In this case the contents change continuously and the sd2snes will not save every change in order to reduce wear on the card but instead save the SRAM contents periodically every 15-20 seconds. **The red LED (Write) will be permanently on if such a case is detected.**

[Source](https://sd2snes.de/blog/faq) (A10, 6th bullet point)

kscorpionx, 2024-08-15:
> [Source](https://sd2snes.de/blog/faq) (A10, 6th bullet point)

Thanks for the answer.

Ramsis-SNES, 2024-08-26:
You're welcome.

Just so you know, you can (and should) close this issue yourself by clicking on the appropriate button down below. :-)

closed 2024-09-01 by kscorpionx