mrehkopf/sd2snes #256: FX Pak Pro does not save system settings

Ronchande, 2024-08-01:
Hi. I have a weird problem with my new SNES FXPak Pro (newest FW). 
It works just fine, saving savegames, savestates and the cfg files like favorites or recent games. The clock also runs fine, it just does not want to save my configuration settings, it always resets to default after every restart.
The config yml file is also empty and has 0 kb. What is the problem? Can I download a „filled out" configuration file somewhere which I can copy on the SD card? Thx. Ron

Ronchande, 2024-08-01:
Tried another sd card and it wrote a 4kb config file. Very strange as all other files could be written. But anyway, solved.

mrehkopf, 2024-08-02:
Strange indeed, maybe something was up with the file system... I'd probably have suggested reformatting the card.
Did you see any periodic flashing from the red LED before changing the SD card?

Ronchande, 2024-08-02:
Is there a LED ? 😅 I do not see one? Made me wonder anyway as there is a led light setting but I see no led? I have a solid grey cartridge, maybe it's just visible in transparent cartridges?

mrehkopf, 2024-08-02:
Ah yeah, there are LEDs soldered on the top left side of the board.
Fun fact: My original *original* design (before mass production) had the bulb-type LEDs soldered on with spacers so they would even poke out of the shell but in turn it required shells to have holes drilled for the LEDs. :sweat_smile:

Ronchande, 2024-08-02:
I don't mind TBH, prefer the original snes cartridge color :)

closed 2024-08-28 by mrehkopf