mrehkopf/sd2snes #258: Super Famicom - Wavy line interference using FXPAK Pro

PJSOConnell, 2024-08-09:
Hi all,

I am using an official FXPAK Pro from everdrive.me using 1.11.0.  
I am using a Super Famicom with a SuperCIC installed, fully recapped, using [this](https://en.retrogamesupply.com/products/power-supply-for-nintendo-super-famicom) power supply and [this](https://www.retrogamingcables.co.uk/packapunch-super-nintendo-entertainment-system-snes-n64-rgb-av-scart-cable-ntsc-gold-scart) RGB cable with an OSSC. 

When using official 

https://github.com/user-attachments/assets/3858335b-e3a9-435e-8b68-edba65a443db

cartridges from all regions I have no issues. However, when using the FXPAK Pro, I get a large amount of wavy lines onscreen, almost resembling interference. 

I have tried this with SuperCIC enabled and disabled, I have replaced the voltage regulator, and I see no difference. I have disabled in-game-hooks. 

I have seen others mention similar issues where people said downgrading to 1.10.2 rectified this issue but I cannot downgrade at all, I just get a black screen. 

Any help would be great.

mrehkopf, 2024-08-12:
I just ordered the power supply and cable to check if I can provoke a similar issue. What model is your Super Famicom?

PJSOConnell, 2024-08-12:
@mrehkopf hey there! My model is the Super Famicom SNS-CPU-APU-01.

Ramsis-SNES, 2024-08-13:
Do you have another 9-10V, 1.3A power supply to try and rule out the one you're currently using as a potential source of trouble?

PJSOConnell, 2024-08-13:
Yes I tested it with a different matching PSU and also Component cables instead of SCART, same result unfortunately.

HLimaVerde, 2025-06-10:
Hi! Any updates about the issue? I have same problem here and no clue how to fix it.

Ramsis-SNES, 2025-06-11:
It's most likely not related to the sd2snes/FXPAK anyway ...

Ramsis-SNES, 2025-06-11:
> > Hi! Any updates about the issue? I have same problem here and no clue how to fix it.
> 
> I had my FXPAK Pro replaced and I no longer have the issue. My console was not responsible.

Great to hear that! :-)