mrehkopf/sd2snes #259: weird SD2SNES clone. Help needed

way5, 2024-08-11:
Hello everyone! 
I am new to this topic so I'd appreciate an expert opinion. A couple of weeks ago I've acquired chinese SD2SNES (which you may see on the photo). 

[image]

It seems to be well made at first glance, however I immediately started having issues with that.

Input data:
- original SFC (US NTSC)
- tested with different PSU 9V(2A), 12V(2A)
- tested with firmwares 1.10.3 (factory), 1.10.3, 1.10.3 (by furious), 1.11.0 (latest, no video)
- tested with various SD cards (32-128GB). the results are the same

System information:

[image]

Issues:
1. The most annoying issues is that I am unable to open directories. Once I am trying to open one the cursor jumps back to the first row in the list. So I've moved sfc roms into SD card root to be able to start them. However...
2. Some games work fine, but most of them having issues at the very beginning of execution or shortly after starting the game. Mostly it just freezes. Sometimes I am able to do IGR though, which returns me to game intro.
3. I am not able to change settings via SD2SNES interface. It's possible to edit `config` directly by hand, but I feel there is nothing changing afterwards. Settings are not applied or having no effect.
4. It says `CIC state: Original or no CIC` which doesn't explain a thing to me.
5. Firmware version 1.11.0 gives me no meaningful video image (see image below). I've tried to follow the steps in this [report](https://github.com/mrehkopf/sd2snes/issues/184), but no luck.

[image]

I've done some reading on Internet about similar issues and possible solutions, but it is not becomes clear to me what is going on to this SD2SNES.

------------- 

**Is there any possible solution of the product is faulty?** 
Please share your opinion.

~Thanks!

mrehkopf, 2024-08-12:
Oh, that's an interesting device I haven't seen before. For issues I think it would be best to contact the dealer/manufacturer :)

Romain80200, 2024-08-12:
Hello, it look like 2 components are missing, R301 and Q301.

way5, 2024-08-12:
> Oh, that's an interesting device I haven't seen before. For issues I think it would be best to contact the dealer/manufacturer :)

I guess this may be sort of a taunt. :) Alright, as I said I am newbie to SFC stuffs. 
I have seen this devices on Internet, and the very first time it is in my hands. And despite its well designed look it seems to be in very beta stage yet, am I correct?
The manufacturer/seller gives no support on that, the only way is to return this device, but I wonder if this is an issue of my SFC maybe, or something else.

Thanks

way5, 2024-08-12:
> Hello, it look like 2 components are missing, R301 and Q301.

Thanks! Is there any schematics which I could refer to?

Ramsis-SNES, 2024-08-13:
Assuming it's not a hardware fault (which is likely, unfortunately), how did you format your SD cards? (Hint: Don't use tools like Rufus and GPT mode, as it will mess things up.)

way5, 2024-08-13:
> Assuming it's not a hardware fault (which is likely, unfortunately), how did you format your SD cards? (Hint: Don't use tools like Rufus and GPT mode, as it will mess things up.)

@Ramsis-SNES, Thank you for sharing your opinion.
Cards were formatted with FAT32 format. I had no issues with that method earlier. Last thing to blame is SD card I think. 
The problem also is that there is no  structured info on this devices what so ever, I was expecting to find schematics at least for earlier versions.
No data on this sort of problems either. Maybe not many people are using it though.

What does this mean when 'no CIC detected'?

Ramsis-SNES, 2024-08-13:
> The problem also is that there is no structured info on this devices what so ever, I was expecting to find schematics at least for earlier versions.

Those have been up [on this very GitHub repository](https://github.com/mrehkopf/sd2snes/tree/develop/pcb/kicad) for 8+ years.

> What does this mean when 'no CIC detected'?

Don't worry about it, it's perfectly normal with an unmodded console; otherwise, it would say, "SuperCIC-enhanced." :-)

way5, 2024-08-13:
@Ramsis-SNES , darn, I am outdated! :) thanks for the information

Which mod do you mention for CIC?

way5, 2024-08-13:
@Ramsis-SNES , so the board on the photo I've posted is basically somewhat "enhanced" version of the official SD2SNES? And schematics in this repo is the only latest updated version for oficial working device?
That was not clear in the beginning. I thought the device version I purchase is the latest upgraded community version.

Which mod do you mention for CIC?

way5, 2024-08-13:
@Ramsis-SNES , so the board on the photo I've posted is basically somewhat "enhanced" version of the official SD2SNES? And schematics in this repo is the only latest updated version for oficial working device?

That was not clear in the beginning. I thought the device version I purchase is the latest upgraded community version and I just don't know where to find the correct schematics.

Ramsis-SNES, 2024-08-14:
> @Ramsis-SNES , so the board on the photo I've posted is basically somewhat "enhanced" version of the official SD2SNES? And schematics in this repo is the only latest updated version for oficial working device?

I honestly don't know the answer to either of those questions.

What you need to understand though is this: The sd2snes/FXpak is an open source project. As such, anyone can make changes to the design, and offer their product commercially. If you as a buyer want the real thing, you need to make sure to only buy from a reputable seller in order to receive a product that works as advertised.

From what you describe, your cart is very likely faulty, and even if it was fully working, it probably isn't exactly built to spec anyway. If I was in your position, I'd return the non-working item, get a refund from the seller, and invest my money in the real deal. :-)

DCSpecial, 2024-08-23:
> No data on this sort of problems either. Maybe not many people are using it though.

FXPak Pro (formely named sd2SNES) is probably _THE_ go to SNES flashcart -> but i would recommend using a genuine one  produced by krikkz instead of chinese "knock offs" based on old (open source) designs

TharathielCB, 2024-08-23:
Go and buy one by krikkz. In that case ikari (mrehkopf) would get some money for his great work. I would not give any support on a very bad Chinese mock off if I were the developer ;)

Romain80200, 2024-08-24:
R343 is also missing.
I don't know if it matter.

way5, 2024-08-25:
@Ramsis-SNES, really appreciate your opinion on that matter. 
I've actually returned this device.
Issue solved.

Thanks

closed 2024-08-25 by way5