mrehkopf/sd2snes #260: Super Mario All Stars + Super Mario World, Graphical Glitch Firmware 1.11.0

kscorpionx, 2024-08-15:
This glitch:
https://github.com/MiSTer-devel/SNES_MiSTer/issues/253 
Happens when you fight Ludwin von Koopa when he jumps and rotates in the air.
Reset patch for clock phase is ON.

Ramsis-SNES, 2024-09-04:
I just tested this using my Super Nt and was able to reproduce it on the first try (_Super Mario All Stars + Super Mario World_ US ROM, 60 Hz mode/NTSC setting).

I then tried i with a real _Super Mario World_ US cartridge and couldn't get it to glitch.

Lastly, I tried it with a real _Super Mario All Stars + Super Mario World_ US cart, and lo and behold, it glitched on the very first try again.

[image]

[image]

So, this is obviously not an sd2snes issue, but maybe a Super Nt bug (or one affecting FPGA SNES implementation methods in general -- bear in mind that I haven't tested it on an actual SNES console just yet). Or, it is possibly a bug only affecting _Super Mario All Stars + Super Mario World_ but not just plain _Super Mario World_.

One more down. ;-)

kscorpionx, 2024-09-06:
> I just tested this using my Super Nt and was able to reproduce it on the first try (_Super Mario All Stars + Super Mario World_ US ROM, 60 Hz mode/NTSC setting).
> 
> I then tried it with a real _Super Mario World_ US cartridge and couldn't get it to glitch.
> 
> Lastly, I tried it with a real _Super Mario All Stars + Super Mario World_ US cart, and lo and behold, it glitched on the very first try again.
> 
> ![vlcsnap-2024-09-04-20h28m31s994](https://private-user-images.githubusercontent.com/11965591/364501221-130aa707-6ab9-41fc-8a46-0b539b3a4f01.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjU2MzUzMjcsIm5iZiI6MTcyNTYzNTAyNywicGF0aCI6Ii8xMTk2NTU5MS8zNjQ1MDEyMjEtMTMwYWE3MDctNmFiOS00MWZjLThhNDYtMGI1MzliM2E0ZjAxLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA5MDYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwOTA2VDE1MDM0N1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTc4M2Q3MTExYjY2ZTJkZmUwZWM4OWIyMGRiYWJjMjEzNWU1NGY3OGQ5MmY3MGVmOTY3MmZhZjAyOTIyM2Q3MWQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.YchPPZF7TauVCBNEYnXbLNXZkLIkPk4yZ5q8F8oMEz0)
> 
> ![vlcsnap-2024-09-04-20h29m23s430](https://private-user-images.githubusercontent.com/11965591/364501679-d850a0cb-2d77-46f3-9704-b73d555eecad.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjU2MzUzMjcsIm5iZiI6MTcyNTYzNTAyNywicGF0aCI6Ii8xMTk2NTU5MS8zNjQ1MDE2NzktZDg1MGEwY2ItMmQ3Ny00NmYzLTk3MDQtYjczZDU1NWVlY2FkLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA5MDYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwOTA2VDE1MDM0N1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTkxMzA4MDJhYzM4MmU1ZjkwODMzNDI1YThlODAwNDU3YzFiOTZkZjU3ZTM5MGY4ZTE3ZTRiZDIxNzAwNDViOWEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.BtrSYXw59HIMhq28ZN64EeSNKmNVbawMufKM2o6x5-g)
> 
> So, this is obviously not an sd2snes issue, but maybe a Super Nt bug (or one affecting FPGA SNES implementation methods in general -- bear in mind that I haven't tested it on an actual SNES console just yet). Or, it is possibly a game software bug that only exists in _Super Mario All Stars + Super Mario World_ but not just plain _Super Mario World_.
> 
> One more down. ;-)

Thanks for the response, and yes, I think this is a technical problem that occurs with that version of Super Mario World, on the Super Mario All Star cartridge. I will try to get my original copy that was left at my parents' house. And I'll try it with real hardware and software.

kscorpionx, 2026-09-27:
I have confirmmed  this on mister fpga this issue doesnt  happen on real hardware, needs  to  be fixed on sd2snes

Ramsis-SNES, 2026-09-27:
@mrehkopf As this is demonstrably _**not**_ an sd2snes/FXPAK issue, you might want to close this.

mrehkopf, 2026-09-27:
Thanks for reminding me of this issue @Ramsis-SNES :)
I think this is linked to newly found PPU register latching timing issues with emulators and/or FPGA implementations of the *SNES*, not the cartridge; as such it isn't fixable within the scope of the cartridge.
https://github.com/snes9xgit/snes9x/issues/1101
(Hint: MiSTer FPGA and Super Nt don't really qualify as "real hardware" baselines in this regard.)

closed 2026-09-27 by mrehkopf

Ramsis-SNES, 2026-09-28:
@mrehkopf Any time :-)

Just for the record (and as you may not have seen my last night edit of a [previous reply](https://github.com/mrehkopf/sd2snes/issues/260#issuecomment-2329747611)), this specific glitch does indeed happen on real hardware¹, i.e. on an original SNES. I wonder what the MiSTer crew actually did to "fix" it, but that's another subject entirely. :D

¹ To me at least, the term "real hardware" always refers to console hardware from the original manufacturer, not modern recreations of any kind.

mrehkopf, 2026-09-28:
Oh I see! Well, maybe it's just another of the few differences the 1CHIP introduces, or another CPU-vs-PPU clock phase hit-and-miss thing (although "Reset patch for clock phase" in the In-game Settings menu should take care of that). Either way it does look like a corner case even for real hardware.
Fully agree on your footnote regarding real hardware. ;)