mrehkopf/sd2snes #262: Super FX screen flicker

Sunlitspace542, 2024-09-13:
I am using an SD2SNES Mk. II Rev. J, running this firmware: https://github.com/mrehkopf/sd2snes/files/10765998/issue186.zip
In some Super FX romhacks, the screen can flash solid black for a frame. Booting the ROM over USB with QUSB2SNES fixes this issue.
Apologies for the blurry video, but it should be enough to demonstrate the issue.

https://github.com/user-attachments/assets/44541d40-f0da-4601-bb29-24975aa16371

mrehkopf, 2024-09-13:
Can you check if you have the brightness limit or 1chip transient fix enabled in configuration->in-game settings? They don't work reliably with every game due to differences in how each game programs the blanking/brightness register.
It probably works using qusb2snes because it doesn't set all features when loading a ROM.

Sunlitspace542, 2024-09-13:
> Can you check if you have the brightness limit or 1chip transient fix enabled in configuration->in-game settings?

Just checked, and I don't.

r0g1nh0, 2024-09-13:
With firmware issue186, very thin vertical lines also appear in the Super Mario RPG game and the graphics are of lower quality in others.

Sunlitspace542, 2024-09-13:
The footage is of Star Fox Next Gen, I am on the development team. This one has the worst case of it, though it does happen a bit on the title screen of Star Fox EX.

mrehkopf, 2024-09-13:
Oh, totally missed that fact ^^; The in-game hook is also a possible cause for problems. I am not aware of any issues with Star Fox, however that may or may not be true for ROM hacks depending on the modifications made...

Sunlitspace542, 2024-09-13:
I don't have in-game hooks enabled either. I also haven't been able to reproduce it on any particularly accurate emulators.

Sunlitspace542, 2024-09-13:
I just tried enabling in-game hooks and the flickering is gone.
I enabled in-game hooks, and the associated options "in-game buttons" and "initial holdoff" are disabled. All other in-game settings are disabled.

mrehkopf, 2024-09-13:
Is there a patch available publicly? I'd like to check it out. (Though I'll be gone over the weekend)
I assume that it doesn't happen on a vanilla ROM? (It doesn't for me)

Sunlitspace542, 2024-09-13:
It's still in very early development at the moment.
I haven't had it happen with vanilla Star Fox or Star Fox 2.

Sunlitspace542, 2024-09-13:
Anyway though, as I said enabling in-game hooks is only a partial fix, as it only fixes flicker in the part I showed in the video.
Booting over USB eliminates all flicker.

mrehkopf, 2024-09-13:
Right now I am leaning towards the issue being a "real hardware" issue rather than an sd2snes/FXPAK issue, with the in-game hook being able to mask it somehow (it has an impact on NMI execution timing). Why it works via USB remains a mystery.
Do you have a chance to test it on a SuperFX donor cart with EEPROM/Flash attached?

Sunlitspace542, 2024-09-13:
> Do you have a chance to test it on a SuperFX donor cart with EEPROM/Flash attached?
I don't, but I know someone who can.

Ramsis-SNES, 2024-09-14:
Just for the record, I tried StarFox EX v1.04.3 (both the normal and the FastROM patch) with my sd2snes Mk. II rev. H, firmware v1.11.0/issue186 on an Analogue Super Nt, and there are no glitches/flickering on either the intro sequence or the title screen whatsoever.

Also, I do own a homebrew GSU-2 flash cartridge, but programming the thing is rather tedious, so ... yeah.

[image]

[image]

[image]

[image]

Sunlitspace542, 2024-09-14:
I tried SFEX 1.10.04 (current public release) and it flickers on the title screen. it's not very frequent though, so it can be easily missed.

Sunlitspace542, 2024-09-17:
> Right now I am leaning towards the issue being a "real hardware" issue rather than an sd2snes/FXPAK issue, with the in-game hook being able to mask it somehow (it has an impact on NMI execution [...]

I just got the video back for the hardware test, no flickering at all.

https://github.com/user-attachments/assets/ca0bc7fd-758c-4d80-ae8c-ac2c77b0072f

Ramsis-SNES, 2024-09-17:
I was able to reproduce the flickering on the title screen of SFEX 1.10.04 tonight. The only thing I changed from the setup mentioned in my previous post was to use a real (2-PPU) PAL SNES (SuperCIC-modded, 60 Hz) hooked up to my Sony PVM-14M2E.

The flickering only appeared after power-cycling and reloading the ROM _twice_, though. Also, it was _extremely_ sporadic. From the looks of it, it could very well be an issue of the hack itself. (It shows garbled graphics for a split second right at the start, even before the kandowontu message, so it doesn't seem to be too polished as of yet.)

I might consider writing the ROM to my GSU-2 flash cart (sigh) to really get to the bottom of this ...

Sunlitspace542, 2024-09-21:
> The garbled graphics also happens with vanilla Starfox v1.2 as well as the Competition ROM, but not with most other SuperFX games. Also, it only ever happens right after loading the game from the menu, but not e.g. after pressing the Reset button when the game was already running. 

I believe that just has to do with how Star Fox clears VRAM on first startup. Star Fox 2 runs on the same engine and it doesn't do it, presumably because VRAM gets cleared properly on boot in that game.