mrehkopf/sd2snes #263: 256mb S-DD1 Rom support

souldbminersmwc, 2024-10-25:
Heya!
I'm making a hack of SMW, and i need to use a large S-DD1 rom. These are currently unsupprorted, even though they SHOULD run perfectly fine on real hardware

mrehkopf, 2024-10-25:
I'm afraid that 256Mbits is out of reach, there is only 128Mbits of RAM on the sd2snes/fxpak.

souldbminersmwc, 2024-10-27:
Then at least 128mb should be supported

souldbminersmwc, 2024-10-27:
And i mean 256 MegaBYTES (Yes sdd1 can go that far)

Ramsis-SNES, 2024-10-27:
This seems oddly familiar. Like ... something I read somewhere close to 15 years ago.

Oh yeah, [now I remember.](https://forums.nesdev.org/viewtopic.php?t=5767) :-) Back then, it was a cool idea, and very impressive.

In this day and age, though? We've had a complete and working MSU-1 design offering even _more_ ROM space (like, 4GB IIRC) for _years_ now, which has rendered the S-DD1-based approach completely obsolete.

Please close this.

mrehkopf, 2024-10-27:
Oh dear 😅 Is there any documentation on that? I found a pinout showing 2 ROM select lines and 21 (maybe 22) address lines, amounting to a total of 16MB addressable space.

mrehkopf, 2024-10-27:
Oops, missed @Ramsis-SNES 's reply in the meantime. ;)  I was referring to this one: https://github.com/mrehkopf/sd2snes/issues/263#issuecomment-2440113953

Ramsis-SNES, 2024-10-27:
@mrehkopf He probably got the number of 256 MB from [this post](https://forums.nesdev.org/viewtopic.php?p=53008#p53008), neglecting the fact that Near was talking about a possible change to the S-DD1 software implementation in bsnes, not actual hardware specs. ;-)

mrehkopf, 2024-10-27:
Hmm, possibly. I guess the internal registers would support such storage sizes. An obvious advantage of S-DD1 (SA-1 too) is zero-latency random access. MSU1 forces you to wait a few ms for the SD Card every time you need to read non-linearly. (I wish emulators were more faithful in that regard, they usually just stop emulation and prepare data synchronously, software won't ever see a busy flag set...)

Ramsis-SNES, 2024-10-27:
> An obvious advantage of S-DD1 (SA-1 too) is zero-latency random access.

True. Then again, what other purpose than video (which MSU1 handles perfectly well) would OP need 256 MB of ROM space for? In a _Super Mario World_ hack, of all things?

Anyway, I'll be more than happy myself to see something cool implemented within FXPAK hardware limitations (so no 256 MB support obviously) ... Feel free to let me know if you need any of smkdan's video ROMs for research purposes. :-) I just tested the "fzeroxvid" one on firmware v1.11.0, but all it produces is a black screen ...

souldbminersmwc, 2024-10-27:
I would like them