mrehkopf/sd2snes #265: Tintin Prisioners of the Sun not working (rapid flashing and broken graphics)

Xylophone4you, 2025-01-09:
I currently am on the latest firmware and I have been having issues with this game. I currently play on an NTSC SNES Mini console, and this game exhibits a lot of flickering and missing objects. It may be perhaps because the game was developed in Europe, but I am not sure if the game can be fixed to run on NTSC consoles even if it is slightly sped up.

mrehkopf, 2025-01-11:
The game is very likely not 60Hz compatible due to video timing. The flash cart is not responsible for fixing the programming of random games :)
However, there seems to be a patch for that game that makes it playable at 60Hz, made by GhostlyDark. https://www.romhacking.net/forum/index.php?topic=23810.0

closed 2025-01-11 by mrehkopf

mrehkopf, 2025-01-12:
Unfortunately the patch only seems to remove the region check but doesn't fix any actual game issues. The game simply spends way too much time in VBlank to read and update VRAM. :(

Xylophone4you, 2025-01-12:
Yeah, I tried using that patch and I wasn't able to get the game running without any major glitches. I hope the SuperCIC will be restocked someday.