mrehkopf/sd2snes #267: Final Fantasy 6 recovery spring graphical glitching

nickretallack, 2025-02-27:
This is what happens when I use a recovery spring:
https://github.com/user-attachments/assets/38bad4fb-552f-4c61-8960-3f399eb2da5d

This was captured using an Elgato HD60 S connected to a Rad2X HDMI adapter.

The graphical glitching is quite strange in that it breaks out of the 4:3 aspect ratio of the image.

Sometimes instead of seeing this glitching, the screen just goes black and there is no sound either, for the duration of the animation.  It comes back afterward.  I can't capture that on my capture device though because my capture device immediately ends the video when that happens.

It's odd since save points should have a similar effect, but they work flawlessly, while this recovery spring does not.
The spring in question is in the caves you escape to on chocobos when Figaro Castle submerges when Kefka is trying to burn it down.

Ramsis-SNES, 2025-03-23:
I suspect this is an issue with your video setup. You might want to try and play around with the brightness limit setting (Main Menu --> Configuration --> In-game settings --> Brightness limit) and report back if/how it affects the Recovery Spring effect.

nickretallack, 2025-04-30:
Good call.  Setting the brightness limit to -1 fixes it.  Is there any way to set this setting but just for this one game?

Ramsis-SNES, 2025-05-02:
I can't answer your questions, but I strongly suggest you change/update/improve your video setup.

Also, please close this issue.

mrehkopf, 2025-05-02:
*If* it's a 1CHIP SNES: these are known to have a somewhat "hot" video output, meaning the output voltage for white is too high. This is also what the brightness limiter is intended to provide a fix for. It is possible to fix in hardware by replacing some parts inside the SNES, which is probably the preferred solution because it doesn't rely on patching register accesses with potential side effects.
I think it should be possible to find some material on that matter by searching for "1chip brightness fix" or "1chip video level fix" or similar.

closed 2025-05-02 by mrehkopf