mrehkopf/sd2snes #268: The Smurfs (Gameboy) is Glitched

magic-jewelry-nerd, 2025-03-01:
Idk if this happens with other Gameboy games, I haven't tested many but for some reason when I try to boot The Smurfs both the sgb enhanced and non sgb enhanced versions it just takes me to the Gameboy bios with the Nintendo logo and just gets stuck there and idk what the problem is. It does this on both the sgb1 and the sgb2. This is the only Gameboy game it's done this to me. I have the latest firmware of sd2snes from 2021. I have a fxpak pro cart, my snes is also American ntsc if you need that info as well.

[image]

magic-jewelry-nerd, 2025-03-03:
I did some research and think it's likely a emulator issue with the Gameboy emulator included with the flash cart/everdrive, seems this game is tricky to emulate as it requires a bug or oversite with the Gameboy that the devs used to make it work that many emulators struggled with in the past. I've found other people mention this same problem on other Gameboy emulators that people were developing that it just hangs on the bios likely cause of this. So guess whoever made the emulator on this flash cart/everdrive will need to figure out how to emulate this game.

https://gbdev.gg8.se/wiki/articles/Tricky-to-emulate_games

mrehkopf, 2025-03-04:
Wow that's a pretty evil bug, and an even more evil one to exploit like that :D So basically it executes 76 f0 f0 8c instead of 76 f0 8c because the byte after 76 (halt) is fetched twice.
Thanks for pointing it out, for now I'll put it on my todo heap but will keep track of it.

magic-jewelry-nerd, 2025-03-05:
Yea that's basically what it is, I really don't know why they did this for this random licensed game, it's very evil for sure when making Gameboy emulators, defiantly recommend checking out the other games in that Gameboy list to see if there's issues with those as well. though defiantly ignore the ones that are gbc only since the sgb is basically a og Gameboy.

magic-jewelry-nerd, 2025-06-18:
Haven't heard anything for a while have you guys been busy with other stuff just curious if there's been a update to fix this game on the Gameboy emulator included with this everdrive/flash cart, I had stuff planned to do but it all got canceled so I'm back to being bored and wondering about this again if it's been fixed and the next firmware will have it patched whenever that will come out. Sorry if this message is annoying I'm just super impatient when it comes to fixing bugs.