mrehkopf/sd2snes #269: Mode 7 is messed up

nickretallack, 2025-05-01:
https://imgur.com/a/iP9nWvd

Github image uploads seem to be broken right now, so here's an imgur link.

Mode 7 graphics aren't working on my Fx Pak Pro.  Many scan lines of the mode 7 graphics are randomly not drawn each frame, allowing the background to be seen instead.  So far I've seen the problem affect Mario Kart, Runsaber, even The Flintstones: The Treasure of Sierra Madrock.

I've tried disabling many settings but nothing seems to change this.

I also noticed that in Faceball 2000 the maze walls often don't render.

I'm using the latest firmware (1.11.0) and I've tried disabling settings like in game hooks, but nothing seems to affect this.

Ramsis-SNES, 2025-05-02:
What cart revision do you have? (Something like "revX" should be printed on the PCB somewhere.)

mrehkopf, 2025-05-02:
That's kinda weird. Are you sure your SNES is ok? Do you happen to have any original cartridges of games where Mode 7 is failing on the FXPAK Pro so you could cross-check?

closed 2026-05-16 by mrehkopf